realsenseai/librealsense #5820: TX2 Realsense-view show error:RS2_USB_STATUS_ACCESS

weseu, 2020-02-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR305  | 
| Firmware Version                   |  2.22.0  | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA TX2  |

### Issue Description
when I open realsense-viewer,I get the following massage.
UNKNOWN in rs2_create_device(info_list:0x55ca266480. index:0);command transfer failed to execute bulk transfer. error:RS2_USB_STATUS_ACCESS

By the way. If I use D435i ,it can work. 
Please help me , thank you.

weseu, 2020-02-10:
by the way, jetpack version is 4.2

dorodnic, 2020-02-10:
Usually this is related to udev rules installation script - depends on the method you installed the sdk (Debian / from source?)

weseu, 2020-02-11:
I install the librealsense from  " https://github.com/Tengyun-Mo/buildLibrealsense2TX2 ".
if it is error at udev rules installation script. how to fix it?

weseu, 2020-02-12:
I fix it by running "sudo apt-get install utils".

thank you.

closed 2020-02-12 by weseu