realsenseai/librealsense #5822: filter pointcloud from binary image

impet14, 2020-02-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| Language                            |  {python/}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi guys,
Does anyone ever filter pointcloud from binary image from color image or depth image?
I am trying to filter pointclould for the object I got the mask but I could not find the solution.
Please suggest me. 
Thank you in advance.

dorodnic, 2020-02-10:
Hi @impet14 
It's not very clear what type of binary format you mean. We strongly recommend using built-in ROS-bag playback and record mechanism to serialise and deserialise frames. There are many other ways to store / load data, but if you chose a different format you are on your own.

impet14, 2020-02-11:
Hi @dorodnic 
Thank you very much for your answer. I understand the step to implement by rosbag because of your suggestion. 
As we would like to do semantic segmentation from color image. Now we could get the mask image of object as a binary image however we would like to filter the pointcloud to get only the poindcloud of the object using mask image we got.

kafan1986, 2020-02-11:
@impet14 
It is easy. Create the pointcloud for only the colour pixel obtained after applying the mask. If you are  currently creating point cloud for the entire image and only need this part, then this approach will save you computation.

Step 1: rs2_project_color_pixel_to_depth_pixel to bring colour pixel to depth pixel, one at a time.
Step 2: rs2_deproject_pixel_to_point to get the point from depth frame

I can post additional code if you need some more guidance.

impet14, 2020-02-12:
Hi @kafan1986 
Thank you for your suggestion.
May I see additional code to be my guide?

kafan1986, 2020-02-12:
> Hi @kafan1986
> Thank you for your suggestion.
> May I see additional code to be my guide?

Look at this https://github.com/IntelRealSense/librealsense/issues/5440#issuecomment-566593866 and https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

RealSenseCustomerSupport, 2020-02-18:
----------------------------------------------

@impet14 Any other questions about this? Thanks!

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@impet14 Will close this if no other questions. Thanks!

closed 2020-03-04 by RealSenseCustomerSupport