realsenseai/librealsense #5824: How to get depth data from raw file?

BeatrizAMartins, 2020-02-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430  depth module | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.22.0 | 
| Language                            |  MATLAB        | 

### Issue Description
I'm using Intel RealSense Depth Module D430. And I have a raw file of a frame that I got from IntelRealsense Viewer and I would like to import that file into MATLAB and get the depth values at one point.  Is this possible and if yes how can I get this in MATLAB?

MartyG-RealSense, 2020-02-11:
Files can be loaded into MATLAB with the **imread** instruction.  I recommend googling for 'matlab imread raw file''.  Here is an example site from such a search:

https://uk.mathworks.com/matlabcentral/fileexchange/66927-read-raw-camera-images

BeatrizAMartins, 2020-02-11:
@MartyG-RealSense thank you! But how can I access to depth data at one point from the raw file?

MartyG-RealSense, 2020-02-11:
Do you mean that you would like to be able to get the depth from a single coordinate in the file, please?

BeatrizAMartins, 2020-02-11:
Yes, I would like to get the depth in a single coordinate from the raw file. Is it possible? Thank you!

MartyG-RealSense, 2020-02-11:
I researched your question carefully.  The link below seemed to be one of the most useful ones for getting depth data from the .raw file:

https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-416814331

BeatrizAMartins, 2020-02-11:
Thank you for the link! Do you know if it's possible to do it in Matlab?

MartyG-RealSense, 2020-02-11:
Most of the references related to getting raw coordinates in MATLAB seem to be related to Kinect cameras.

I did find some code an Intel support team member provided for reading a RealSense-made .raw file in MATLAB.

**FID = fopen('IntelDepth.raw');**
**DepthImage = fread(FID, [1280, 720], 'uint16')'**;
**fclose(FID);**

Make sure you configure the depth image as z16 (16 bit)

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@BeatrizAMartins Could you please specify how you get the raw file of a frame from RealSenseViewer? Thanks!

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@BeatrizAMartins Could you please update? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-03-11:
----------------------------------------------

@BeatrizAMartins Sorry that we didn't get response from your side for weeks. Will close the issue at this point. Please create new one if you still have questions or issues.  Thanks!

closed 2020-03-11 by RealSenseCustomerSupport