realsenseai/librealsense #5827: Cartographer with camera D435 and ROS2

ninamwa, 2020-02-11:
Hi. I want to use Cartographer together with my **D435 camera and ROS Eloquent, in Ubuntu 18.**
I have been trying to use this tutorial (https://intel.github.io/robot_devkit_doc/pages/rs_slam.html) but get some problems as I do not have the T265 camera. What I have done is to comment out the T265 node, and also change the tracking_frame from "camera_pose_optical_frame" to "base_link" as the first one is related to the T265 camera.

By watching the tf-tree, I can see that a lot of frames are not connected.

[frames.pdf](https://github.com/IntelRealSense/librealsense/files/4184935/frames.pdf)

(even though the T265 node is commented out, the static transform between T265 and D435 is still published, thats why it is in the tf-tree).
What can be seen is that there lacks a connection from odom_frame to the rest and also from the D435 link.
Am I supposed to change "odom_frame" in the code with "odom", and the camera links in the turtlebot URDF with the D435 camera? This would might connect the whole tree, but seems like a lot of changes in the code. From the tutorial, I got the impression that I could just run the commands and that it would be straight forward without changing the code too much, but now I am unsure.

Have anyone tried to do the same and could help me out a bit?
Thank you!

MartyG-RealSense, 2020-02-11:
A few years ago a complete hardware and software tutorial for building a RealSense turtlebot was published.  It was designed for the old RealSense R200 camera and gmapping instead of Cartographer, but the author said that a 400 Series camera and Cartographer could be substituted into the design.

https://software.intel.com/en-us/articles/build-an-autonomous-mobile-robot-with-the-intel-realsense-camera-ros-and-sawr

In its tf_frame, it used **odom**.

[image]

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@ninamwa Did you follow the https://intel.github.io/robot_devkit_doc/pages/rs_slam.html step by step? Is it possible to submit this issue on that project to get more help?

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@ninamwa Sorry we didn't get response from you for weeks. Will close the issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-03-11 by RealSenseCustomerSupport