realsenseai/librealsense #5828: Multi-camera issue: After connecting the sixth camera, its motion frame doesn't arrive

wangroger0801, 2020-02-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-76 and 5.0.0-23                                  | 
| Platform                                 | PC   |
| SDK Version                            |  2.32.1                          | 
| Language                            |  C++                           | 
| Segment			|  Others                     | 

Hardware: i7-9800X, Qudro RTX4000 , Nvidia driver 435.21

### Issue Description
Hello,
We are going to use six or more cameras in our system. On windows10, all demos from sdk are fine without problem.
  
On Ubuntu, different errors start appearing after plug in more cameras.
The main issue which stop us developing is we cam not use the motion frame from the sixth camera.  (In demo rs-multicam, it only show 5 cameras when 6 are plugged in).
Could your team check how we could fix this issue? Thanks!

Monitoring the frame data from the Realsense Viewer.
Testing procedures: 
1. plug in one camera
2. start depth frame,color frame, motion frame
3. read error message
4. plug in the next camera

**First CAM**

```
 11/02 10:56:43,100 WARNING [140697856358144] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
 11/02 10:56:43,881 WARNING [140697466291968] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10.2/2-10.2:1.5/0003:8086:0B3A.0006/HID-SENSOR-200073.1.auto/iio:device0/buffer/enable
```

**Second CAM**

**Third CAM**

```
 11/02 10:58:39,747 ERROR [140698292938496] (types.h:304) get_dev_capabilities: Cannot open '/dev/video17 Last Error: Permission denied
 11/02 10:58:39,749 ERROR [140698292938496] (types.h:304) get_dev_capabilities: Cannot open '/dev/video15 Last Error: Permission denied
 11/02 10:58:39,750 ERROR [140698292938496] (types.h:304) get_dev_capabilities: Cannot open '/dev/video13 Last Error: Permission denied
 11/02 10:58:39,751 ERROR [140698292938496] (types.h:304) get_dev_capabilities: Cannot open '/dev/video16 Last Error: Permission denied
 11/02 10:58:39,751 ERROR [140698292938496] (types.h:304) get_dev_capabilities: Cannot open '/dev/video14 Last Error: Permission denied
 11/02 10:58:39,751 ERROR [140698292938496] (types.h:304) get_dev_capabilities: Cannot open '/dev/video12 Last Error: Permission denied
```

**4th CAM**
```
 11/02 11:00:05,162 ERROR [140698292938496] (backend-v4l2.cpp:598) Metadata node for uvc device: id- /dev/video22
vid- 8086
pid- b3a
mi- 3
unique_id- 6-1-2
path- /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:02.0/0000:1a:00.0/usb6/6-1/6-1:1.3/video4linux/video22
susb specification- 320
metadata node-/dev/video23 was already been assigned 
```
**5th CAM**
``` 
 11/02 11:00:50,402 ERROR [140698292938496] (backend-v4l2.cpp:598) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- b3a
mi- 3
unique_id- 4-1-2
path- /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1:1.3/video4linux/video28
susb specification- 320
[... 129 lines cut ...]
        |__ Port 2: Dev 6, If 0, Class=Hub, Driver=hub/2p, 5000M
            |__ Port 2: Dev 7, If 4, Class=Video, Driver=uvcvideo, 5000M
            |__ Port 2: Dev 7, If 2, Class=Video, Driver=uvcvideo, 5000M
            |__ Port 2: Dev 7, If 0, Class=Video, Driver=uvcvideo, 5000M
            |__ Port 2: Dev 7, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
            |__ Port 2: Dev 7, If 3, Class=Video, Driver=uvcvideo, 5000M
            |__ Port 2: Dev 7, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 9: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
    |__ Port 10: Dev 3, If 0, Class=Hub, Driver=hub/2p, 5000M
        |__ Port 2: Dev 4, If 5, Class=Human Interface Device, Driver=usbhid, 5000M
        |__ Port 2: Dev 4, If 3, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 4, If 1, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 4, If 4, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 4, If 2, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 4, If 0, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 5: Dev 6, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 2: Dev 7, If 0, Class=Hub, Driver=hub/2p, 480M
    |__ Port 10: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 1: Dev 5, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 11: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 11: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M
```

schmiran, 2020-02-12:
Hello, I have the same problem. 

I am running Ubuntu 18.04.4 LTS - 5.3.0-28-generic. The installation of realsense 2.32.1 via the Debian package worked without error messages. Also the kernel patch:

```
version:        1.1.2.realsense-1.3.13
srcversion:     3DD05991EAABF8F5DE4E188
```

Whether I use the sensors inside ROS or just the realsense-viewer. Starting with the 4th sensor I get the error message:

```
12/02 11:14:27,013 ERROR [140128714475264] (backend-v4l2.cpp:598) Metadata node for uvc device: id- /dev/video20
vid- 8086
pid- ad3
mi- 3
unique_id- 6-1-14
path- /sys/devices/pci0000:00/0000:00:1c.0/0000:04:00.0/0000:05:02.0/0000:2c:00.0/usb6/6-1/6-1:1.3/video4linux/video20
susb specification- 320
metadata node-/dev/video21 was already been assigned
```

I have exchanged the cables, as well as the ports, the error always comes with the 4th connected sensor regardless of the configuration. 

I also have the problem that if I connected the sensors via a Thunderbolt 3 hub like before, there are no more frames when the sensor is turned off and on again. Operation via the hub was no problem before. 

Despite the error message starting with the 4th sensor, I get data in realsense-viewer as long as I do not run the sensors over the hub. What does not work is to work with the sensors in ROS. The nodes that subscribe to the corresponding channels of the sensors don't get any data anymore. This was running without problems before and no changes were made at this point by me.

In about one of 30 attempts my ROS configuration works and I can work with the sensor data as usual. Why it sometimes works and sometimes not is not reproducible for me. If it does not work I either get the following error message from t265:

`[ WARN] [1581501726.513505941]: Error extracting usb port from device with physical ID: 2-1-8`

Or I get an error message that the camera_manager has been disabled. Unfortunately I don't get the error reproduced right now (if it appears again, I'll add this message here).

I have also noticed that the physical port on the T265 is e.g. `2-1-8`, whereas on the other sensors (2x D435, 1x D415) it is like: `sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video4`

With the t265 I also noticed that the serial number has changed since realsense 2.32.1. Four zeros were added at the front.

I hope you can help me, I'm a bit confused why my system doesn't work as usual anymore.
Greetings

wangroger0801, 2020-02-12:
Hello @schmiran ,

Today we rebuilt the library from development branch (Latest commit d3980b2). 
New library v2.32.1 and new firmware 5.12.02.100
It seems that we get rid of the error messages saying `metadata node-/dev/video23 was already been assigned `. Is it something you can do a try-out?

**However the motion frame is still not arrived from the sixth cameras** 
New error messages when I switch on/off Motion Module on the camera with issue.

```
 12/02 14:30:14,929 WARNING [140421577557760] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.5/0003:8086:0B3A.0001/HID-SENSOR-200073.1.auto/iio:device0/buffer/enable
 12/02 14:30:23,727 WARNING [140419912423168] (backend-hid.h:68) write_fs_attribute Could not change gyro_3d-dev11 to gyro_3d-dev1 : path /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/trigger/current_trigger
 12/02 14:30:23,730 WARNING [140419920815872] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:23,730 WARNING [140419920815872] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:24,571 WARNING [140419920815872] (backend-hid.cpp:681) HID set_power 0 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:25,710 WARNING [140419895637760] (backend-hid.h:68) write_fs_attribute Could not change  to gyro_3d-dev1 : path /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/trigger/current_trigger
 12/02 14:30:25,715 WARNING [140419904030464] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:25,715 WARNING [140419904030464] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:30,719 WARNING [140419912423168] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:30,720 WARNING [140419887245056] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:32,982 WARNING [140419904030464] (backend-hid.cpp:681) HID set_power 0 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:35,499 WARNING [140419887245056] (backend-hid.h:68) write_fs_attribute Could not change  to gyro_3d-dev1 : path /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/trigger/current_trigger
 12/02 14:30:35,503 WARNING [140419912423168] (backend-hid.h:68) write_fs_attribute Could not change 0 to 1 : path /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:35,503 WARNING [140419912423168] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:30:40,507 WARNING [140419920815872] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:40,507 WARNING [140419895637760] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:45,511 WARNING [140419895637760] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:45,511 WARNING [140419920815872] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:50,515 WARNING [140419920815872] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:50,515 WARNING [140419895637760] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:55,519 WARNING [140419895637760] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:55,519 WARNING [140419920815872] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 12/02 14:30:57,315 WARNING [140419912423168] (backend-hid.cpp:681) HID set_power 0 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
[... cut ...]
 12/02 14:30:59,316 WARNING [140419920815872] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
 12/02 14:31:00,299 WARNING [140419920815872] (backend-hid.cpp:681) HID set_power 0 failed for /sys/devices/pci0000:16/0000:16:00.0/0000:17:00.0/0000:18:01.0/0000:19:00.0/usb4/4-1/4-1.2/4-1.2.2/4-1.2.2:1.5/0003:8086:0B3A.000B/HID-SENSOR-200076.22.auto/iio:device11/buffer/enable
```

wangroger0801, 2020-02-12:
Well... I found out the same error `metadata node-/dev/video23 was already been assigned` after I enable the advanced mode in realsense viewer

schmiran, 2020-02-13:
Hey @wangroger0801 ,

so installing the development branch didn't help after all? I'll try it this afternoon and let you know. 

Regarding your error message with the frames:
> (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds

I get the same error message when I run sensors across the Hub. Even on the first sensor. If I connect them directly to the system, I don't get that error message.

schmiran, 2020-02-13:
For the problem with the wrong physical ID I found this https://github.com/IntelRealSense/realsense-ros/issues/1051#issuecomment-576942007 and will test it later.

schmiran, 2020-02-17:
Hey @wangroger0801 , 

I have now also tested the developer branch, without changes. The other hint about the physical ID couldn't help me either.

Do you have new information? Or maybe @RealSenseCustomerSupport  has an idea what causes the error messages?

wangroger0801, 2020-02-19:
Hello @schmiran ,

Thanks for your reply. We did a new test. It fixed the motion frame not arrived problem. But more errors pop out. We still can't using 6 cameras. I am sure it not hub related it works fine on Windows. 
We add -DFORCE_RSUSB_BACKEND=true to the SDK cmake. In realsense-viewer , there is no motion frame error when 6 cameras is connected. But there seems an up-limit of video frame data. No video frames received from the third 720p@30Hz RGB frames. To enable all 6 RGB and depth frames plus 6 infrared frames we have to set the resolution to 360p or even lower on some cameras. 
Although we did enable all the frames, our multi-cam program failed to start. It reported  `failed to set power state` when there are more than 1 camera connected.

@RealSenseCustomerSupport  any idea how we can fix this issue?
Thanks.

schmiran, 2020-03-17:
With the new Realsense version it works again. I still have the problem, that if I switch the sensor off and on again, no more frames are coming but this can be solved with an `initial reset` in the launchfile.

I hope the new version has helped you too. It's a pity that there was no support for this topic.

Greetings

tispratik, 2020-03-24:
We're facing the same issue. When we connect the 2nd camera, we get "Cannot set power state" error.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00| 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-88-generic                                         | 
| Platform                                 |    - |
| SDK Version                            |  2.33.1                          | 
| Language                            |  python                         | 
| Segment			|  Robot                          |

tispratik, 2020-03-25:
I can't seem to find documentation regarding low power mode https://github.com/IntelRealSense/librealsense/pull/3654/commits/672b33b7cd7e40fdd54f479f1278fe903e4d2985

In our 2nd camera, we need to run only RGB stream. Can we run it in low power mode to get around the cannot set power state issue?

tispratik, 2020-03-25:
Interestingly when I run the RGB only streaming camera first and then the first camera which does RGB + Depth streaming, both the cameras run fine. Seems like when i run the RGB+Depth camera first, it takes up more power than it needs and then does not leave any for the RGB only streaming Realsense camera.

Need some way to limit the resources consumed by each Realsense camera.

tispratik, 2020-03-25:
```cpp
void rs_uvc_device::set_power_state(power_state state)
{
    _action_dispatcher.invoke_and_wait([&, this](dispatcher::cancellable_timer c)
    {
        if(state != _power_state)
        {
            switch(state)
            {
                case D0:
                    _messenger = _usb_device->open(_info.mi);
                    if (_messenger)
                    {
                        listen_to_interrupts();
                        _power_state = D0;
                    }
                    break;
                case D3:
                    if(_messenger)
                    {
                        close_uvc_device();                            
                        _messenger.reset();
                    }
                    _power_state = D3;
                    break;
            }
        }
    }, [this, state](){ return state == _power_state; });

    if(state != _power_state)
        throw std::runtime_error("failed to set power state");
}
```

The 2nd camera (RGB only) is unable to open the USB device for some reason when the 1st camera (RGB + Depth) is fired up first.

From the above function, seems like D0 is the power state.

tispratik, 2020-03-25:
@MartyG-RealSense any thoughts on this one?

MartyG-RealSense, 2020-03-25:
@tispratik   This link may give you a useful lead:

https://github.com/IntelRealSense/librealsense/issues/5782#issuecomment-582077781

tispratik, 2020-03-25:
Thanks Marty. I had seen that, but does not seem to apply to our case. Our case is 1 camera streaming depth+RGB connected to USB 3.1 port and another camera streaming only RGB stream connected to a USB 3.0 4Amps hub which is in turn connected to another USB 3.0 port on the motherboard.

We have the exact same issue on 3 sets of our machines, so it is a reproducible problem. Interesting to note is that when we run the RGB only camera first, then the RGB+depth camera, both play along nicely on power. But if we run the RGB+depth camera first, then the RGB only camera gives the error of "cannot set power state".

MartyG-RealSense, 2020-03-25:
It sounds as though one camera is connected to the hub and one is connected directly to the computing device.  If that is the case, connecting both to the hub should allow both cameras to be handled by the hub's USB controllers.

tispratik, 2020-03-25:
Yes, we tried connecting both cameras to the hub as well as both cameras directly to the motherboard. We see the same issue.

tispratik, 2020-03-25:
Update: 

Even though we could run the two cameras by starting the RGB camera first and then the RGB+depth one, the RGB camera dies after some time. Following is what i see in dmesg.

```
[Wed Mar 25 02:43:11 2020] perf: interrupt took too long (2504 > 2500), lowering kernel.perf_event_max_sample_rate to 79750
[Wed Mar 25 05:55:53 2020] perf: interrupt took too long (3149 > 3130), lowering kernel.perf_event_max_sample_rate to 63500
[Wed Mar 25 18:47:15 2020] usb 2-3: Process 9334 (python) called USBDEVFS_CLEAR_HALT for active endpoint 0x84
[Wed Mar 25 18:47:24 2020] usb 2-3: Process 9334 (python) called USBDEVFS_CLEAR_HALT for active endpoint 0x84
```

tispratik, 2020-03-25:
A little research on the error points me to https://www.spinics.net/lists/linux-usb/msg102772.html
The Kernel is logging a warning that USBDEVFS_CLEAR_HALT was called and it is wrong to call that function when URBs are queued.

The `usb_clear_halt()` function is being called from librealsense library to reset the endpoint which I am not sure why the camera is trying to do. Seems like the URBs need to be cleared before resetting the end point. This seems like a bug to me @MartyG-RealSense.

```cpp
usb_status usb_messenger_libusb::reset_endpoint(const rs_usb_endpoint& endpoint, uint32_t timeout_ms)
{
    int ep = endpoint->get_address();               
    auto sts = libusb_clear_halt(_handle->get(), ep);
    if(sts < 0)
    {
        std::string strerr = strerror(errno);
        LOG_WARNING("reset_endpoint returned error, index: " << ep << ", error: " << strerr << ", number: " << int(errno));
        return libusb_status_to_rs(sts);
    }
    return RS2_USB_STATUS_SUCCESS;
}
```

MartyG-RealSense, 2020-03-25:
@tispratik  These messages are a Linux kernel issue rather than a Librealsense error.  A cause of the messages you experienced in the post above can apparently be high CPU load.

If you are having kernel and patching problems then maybe you could consider the **backend** SDK installation method that does not rely on kernel versions and patching.  You will need an internet connection though.

https://support.intelrealsense.com/hc/en-us/community/posts/360037657234/comments/360009799533

tispratik, 2020-03-26:
Thanks Marty. Do you mean to say that we should set the flag `-DFORCE_RSUSB_BACKEND=true`.. our current flags are 

```-DBUILD_EXAMPLES=true -DENFORCE_METADATA=true -DBUILD_PYTHON_BINDINGS=true -DFORCE_LIBUVC=true -DBUILD_WITH_CUDA=true -DIMPORT_DEPTH_CAM_FW=false```

MartyG-RealSense, 2020-03-26:
Yes, set **DFORCE_RSUSB_BACKEND=true**

wangroger0801, 2020-03-26:
@MartyG-RealSense Thanks for your support.
Our initial error was not related to `set power state`  and the error occurred only after the sixth camera was connected. 
The realsense viewer said 
`iio_hid_sensor: Frames didn't arrived within 5 seconds`

I also tired `-DFORCE_RSUSB_BACKEND=true`. But it did not work, as I mentioned before.

MartyG-RealSense, 2020-03-26:
My understanding is that a mains-powered USB 3 hub can supply about 12 W of power, and each individual RealSense 400 Series camera requires 2 W.  So I wonder if your sixth camera has insufficient power available.  If each hub has 12 W available, using more than one powered hub may help, putting some of the cameras on the second hub.  In the USB specification, up to five hubs can be connected together (a setup known as '5 deep').

wangroger0801, 2020-03-26:
We have a PCI-E USB adapter which should give sufficient power. And It doesn't explain the reason why it works on windows with the exact configuration.

MartyG-RealSense, 2020-03-26:
A PCI-E USB adapter will still be drawing the power for the port from the computer's power supply unit in the same way that the built-in USB ports do (i.e they are 'passive' ports on the adapter and vulnerable to instabilities like the built-in ports).

Are you using Windows and Ubuntu in dual-boot on the same computer, please?

wangroger0801, 2020-03-26:
It is using PCI-E x4 which should give 25W power to the USB ports. 
We did try windows, ubuntu and archlinux.  Yes in dual-boot mode. Also, the possible secure boot option cause was eliminated.

MartyG-RealSense, 2020-03-26:
The amount of wattage is important but not the only factor.  I have seen cases involving 650 W power supply units on enterprise servers where camera disconnections could still occur.  The stability of the power supplied to the port matters as much as the amount of wattage supplied.  Mains-powered external USB hubs give greater power stability than the internal power supply units of computers.

At this point, I will refer the case internally to a support team manager to see if they can provide advice.   I apologise for the delay in the meantime.

tispratik, 2020-03-27:
I could reliably reproduce the problem with the below script:

```python
import pyrealsense2 as rs
import numpy as np
import cv2

devices = rs.context().query_devices()

pipeline_1 = rs.pipeline()
pipeline_2 = rs.pipeline()

config_1 = rs.config()
config_2 = rs.config()
config_1.enable_device(devices[0].get_info(rs.camera_info.serial_number))
config_2.enable_device(devices[1].get_info(rs.camera_info.serial_number))

config_1.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60)
config_1.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 60)
config_2.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60)
config_2.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 60)

pipeline_1.start(config_1)
pipeline_2.start(config_2)

while True:
    frames_1 = pipeline_1.wait_for_frames()
    frames_2 = pipeline_2.wait_for_frames()
    color_frame_1 = frames_1.get_color_frame()
    color_frame_2 = frames_2.get_color_frame()

    color_image_1 = np.asanyarray(color_frame_1.get_data())
    color_image_2 = np.asanyarray(color_frame_2.get_data())

    cv2.namedWindow('CAM 1', cv2.WINDOW_AUTOSIZE)
    cv2.imshow('CAM 1', color_image_1)

    cv2.namedWindow('CAM 2', cv2.WINDOW_AUTOSIZE)
    cv2.imshow('CAM 2', color_image_2)

    cv2.waitKey(1)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
```

The above script as is may or may not work for you. What is very whacky about the whole issue is that if you swap the order of starting the pipelines 1 and 2......it works!! The order in which we start the two pipelines matters for some unknown reason which the Realsense team needs to dig into.

Try:
```python
pipeline_1.start(config_1)
pipeline_2.start(config_2)
```

vs:

Swap pipelines start:
```python
pipeline_2.start(config_2)
pipeline_1.start(config_1)
```

MartyG-RealSense, 2020-03-27:
@tispratik  Thanks so much for your code.  I do not have a Python development environment set up at the moment to test it myself.  I have contacted a support manager about this case though so it can be looked at.

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@tispratik, thank you for sharing the code. The two pipelines have the same stream profiles (I assume two cameras shares the same firmware). Is there any difference in the physical connection in respect to each device?
@wangroger0801, some pcie raiser card provides a 4 pin power connector to draw extra power from external source which could provide better stability as mentioned as @MartyG-RealSense mentioned before. Just out of curiosity, does only stream 6 motion frames without any depth nor rgb stream work?
Thank you.

tispratik, 2020-03-27:
Following are two more experiments I did on the python (3.6.8) code I provided.

* The issue persists on Ubuntu 16.04 even if I only stream two RGB profiles. No depth streaming.
* The issue does not happen on Windows machine. In Windows, the order of starting the streaming doesn't matter. It works with RGB+depth on both cameras.

tispratik, 2020-03-27:
Based on my testing, I do not think it is a power issue at all. IMO, when trying to starting the 2nd camera, the librealsense library is most likely trying to connect to the 1st camera (which is already running) and hence the "cannot set power state error".

Try the experiment, start 1 camera and try to enumerate the devices through `rs-enumerate-devices --short`, you will see the same "cannot set power state error" on that 1 camera.

tispratik, 2020-03-28:
Info Logs:

```
 28/03 06:36:17,463 INFO [139683223242496] (environment.cpp:85) Found 4 unreachable streams, 4 extrinsics deleted
 28/03 06:36:17,905 INFO [139683223242496] (environment.cpp:85) Found 4 unreachable streams, 4 extrinsics deleted
 28/03 06:36:18,310 INFO [139683223242496] (sensor.cpp:1338) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 28/03 06:36:18,362 INFO [139683223242496] (uvc-streamer.cpp:28) endpoint 130 read buffer size: 615424
 28/03 06:36:18,376 INFO [139683223242496] (sensor.cpp:1338) Request: RGB8 Color, 
Resolved to: YUYV Color, 
 28/03 06:36:18,429 INFO [139683223242496] (uvc-streamer.cpp:28) endpoint 132 read buffer size: 615424
 28/03 06:36:18,446 ERROR [139681815566080] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 28/03 06:36:18,477 INFO [139683223242496] (environment.cpp:85) Found 3 unreachable streams, 1356 extrinsics deleted
 28/03 06:36:18,478 ERROR [139681823958784] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 28/03 06:36:18,514 INFO [139683223242496] (environment.cpp:85) Found 3 unreachable streams, 1356 extrinsics deleted
 28/03 06:36:18,515 ERROR [139681664595712] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 28/03 06:36:18,538 WARNING [139683223242496] (rs.cpp:1718) null pointer passed for argument "profile"
 28/03 06:36:18,584 INFO [139683223242496] (frame-archive.h:215) The user was holding on to 1 frames after stream 0x0x13fc040 stopped
 28/03 06:36:18,670 INFO [139683223242496] (frame-archive.h:225) All frames from stream 0x0x13fc040 are now released by the user
```

Debug logs
```
 28/03 06:33:21,201 DEBUG [140242101139200] (context.cpp:108) Librealsense VERSION: 2.33.1
 28/03 06:33:21,327 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:21,327 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:21,333 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:21,333 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:21,341 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:21,341 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:21,341 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-5-2
 28/03 06:33:21,695 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-5-2
 28/03 06:33:21,721 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 4-1-2
 28/03 06:33:21,729 INFO [140242101139200] (environment.cpp:85) Found 4 unreachable streams, 4 extrinsics deleted
 28/03 06:33:22,092 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 4-1-2
 28/03 06:33:22,122 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:22,122 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:22,122 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-5-2
 28/03 06:33:22,128 INFO [140242101139200] (environment.cpp:85) Found 4 unreachable streams, 4 extrinsics deleted
 28/03 06:33:22,385 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-5-2
 28/03 06:33:22,521 INFO [140242101139200] (sensor.cpp:1338) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 28/03 06:33:22,574 INFO [140242101139200] (uvc-streamer.cpp:28) endpoint 130 read buffer size: 615424
 28/03 06:33:22,586 DEBUG [140242101139200] (global_timestamp_reader.cpp:132) time_diff_keeper::start: _users_count = 1
 28/03 06:33:22,588 INFO [140242101139200] (sensor.cpp:1338) Request: RGB8 Color, 
Resolved to: YUYV Color, 
 28/03 06:33:22,641 INFO [140242101139200] (uvc-streamer.cpp:28) endpoint 132 read buffer size: 615424
 28/03 06:33:22,653 DEBUG [140242101139200] (global_timestamp_reader.cpp:132) time_diff_keeper::start: _users_count = 2
 28/03 06:33:22,654 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:22,654 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:22,654 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-5-2
 28/03 06:33:22,658 ERROR [140240547215104] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 28/03 06:33:22,676 DEBUG [140240971224832] (uvc-streamer.cpp:65) Passing packet to user CB with size 614648
 28/03 06:33:22,681 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:22,681 DEBUG [140242101139200] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 28/03 06:33:22,681 DEBUG [140242101139200] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-5-2
 28/03 06:33:22,689 INFO [140242101139200] (environment.cpp:85) Found 3 unreachable streams, 1356 extrinsics deleted
 28/03 06:33:22,690 ERROR [140240698185472] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 28/03 06:33:22,693 DEBUG [140240971224832] (uvc-streamer.cpp:65) Passing packet to user CB with size 614648
 28/03 06:33:22,698 DEBUG [140240748541696] (global_timestamp_reader.cpp:109) CLinearCoefficients::calc_value: 3.46974e+06 -> 1.58538e+12 with coefs:1, 0, 3.46975e+06, 1.58538e+12
[... cut ...]
 28/03 06:33:22,839 DEBUG [140242101139200] (global_timestamp_reader.cpp:146) time_diff_keeper::stop: stop object.
 28/03 06:33:22,881 DEBUG [140242101139200] (frame-archive.h:148) CallbackFinished,Depth,4,DispatchedAt,1585377202881.854492
 28/03 06:33:22,881 DEBUG [140242101139200] (frame-archive.h:155) Frame Callback [Depth#4] overdue. (Duration: 137.271240ms, FPS: 60, Max Duration: 16ms)
 28/03 06:33:22,881 INFO [140242101139200] (frame-archive.h:225) All frames from stream 0x0x2cbbea0 are now released by the user
```

tispratik, 2020-03-30:
Hey @MartyG-RealSense any update from your team? The python code i provided is very trivial and easy to replicate in C++ if you wish to test it on your computer. Btw, this does not happen on Windows, so you will need a linux based machine like Ubuntu for reproducing the problem.

MartyG-RealSense, 2020-03-30:
@tispratik  It is likely that @RealSenseCustomerSupport who posted a few days ago is the team member referred to this case.

https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-605203998

@RealSenseCustomerSupport  Are you able to offer further input on this case for @tispratik please?   Thanks!

RealSenseCustomerSupport, 2020-03-30:
----------------------------------------------

@tispratik,
The provided script can stream two D435(i) on a Ubuntu 16.04 laptop and I haven't yet experienced the issue starting 2nd camera in the pipeline. Both cameras were connected to a USB type C-A hub.
Just for reference, here is the hardware setup:
Intel® Core™ i5-7360U CPU @ 2.30GHz × 4
Intel® Iris Plus Graphics 640
8GB RAM

Attachment(s):
github-5828_debug.png - https://realsensesupport.intel.com/attachments/token/Ma60uzHWhhRrZWXIMtW30g6zw/?name=github-5828_debug.png

tispratik, 2020-03-31:
Thanks @RealSenseCustomerSupport, I cannot access the script, it says "You are not authorized to see the file. Make sure that you are logged in here.", It does not let me access the file even though I am logged into https://realsensesupport.intel.com.

Also, in your script, have you tried flipping the order of the starting of the pipelines?

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

Actually it's just an attachment of the screen capture of running two cameras with your provided script. :)
Let me upload it here and see if it works:
https://ibb.co/gWHTvN3

I have tried the original then flipped the order of the pipeline start and both worked.

By the way, does multicam example (rs-multicam) work for the two cameras on your side?
https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

tispratik, 2020-03-31:
Interesting to know that it works for you. Can you provide some details on what patches you have applied to the kernel?

The `rs-multicam` also is not working, we are seeing the same error as below.

```
$ rs-multicam 
RealSense error calling rs2_pipeline_start_with_config(pipe:0x275d520, config:0x275d540):
    failed to set power state
```

RealSenseCustomerSupport, 2020-04-01:
----------------------------------------------

I applied the regular `./scripts/patch-realsense-ubuntu-lts.sh`
from https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
Then CMake with Python binding build.
`cmake ../ -DBUILD_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=true`
The system is running on Python 3.5.2

tispratik, 2020-04-01:
Thanks for the details. The only change I see is that we are running on Python 3.6.8.
Don't think though Python is an issue as the `rs-multicam` which is C++ based is also not working for me.

tispratik, 2020-04-02:
> Info Logs:
>
> ```
>  28/03 06:36:17,463 INFO [139683223242496] (environment.cpp:85) Found 4 unreachable streams, 4 extrinsics deleted
>  28/03 06:36:17,905 INFO [139683223242496] (environment.cpp:85) [...]

Is the above debug logs any helpful to nail the issue?

RealSenseCustomerSupport, 2020-04-02:
----------------------------------------------

@tispratik, thank you for the log and Python should not be an issue if C++ rs-multicam example failed to stream the second camera.
I have seen other developers experienced the similar error in the realsense-ros forum which you may find them useful.
https://github.com/IntelRealSense/realsense-ros/issues/1115
or https://github.com/IntelRealSense/realsense-ros/issues/1002

Thank you.

tispratik, 2020-04-02:
Thank you for the links. Looks like in https://github.com/IntelRealSense/realsense-ros/issues/1115, they did apply a fix for that. I believe similar fix needs to be made in librealsense.

longwoo, 2020-04-26:
@MartyG-RealSense @tispratik  
I met the same problem.

I have a 435i and a 415 camera with both firmware 05.12.03.00.

And the cameras both connected to the same usb3.0 hub on Xavier with Jectpack 4.3 .

If using the realsense-viewer, I can see both camera stream , but if using c++ api ， it crashed.

the test code is
```
    rs2::context                          ctx;        // Create librealsense context for managing devices

    std::map<std::string, rs2::colorizer> colorizers; // Declare map from device serial number to colorizer (utility class to convert depth data RGB colorspace)

    std::vector<rs2::pipeline>            pipelines;

    std::cout<<"test0 ==== "<<std::endl;

    // Start a streaming pipe per each connected device
    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        pipe.start(cfg);
        std::cout<<"test1 ==== "<<std::endl;
        pipelines.emplace_back(pipe);
        // Map from each device's serial number to a different colorizer
        colorizers[dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER)] = rs2::colorizer();
    }
```
It gives me 

```
test0 ==== 
test1 ==== 
terminate called after throwing an instance of 'rs2::error'
  what():  failed to set power state
Aborted

```

MartyG-RealSense, 2020-04-26:
@longwoo   If your hub is an "unpowered" hub (one that is connected to the computer and draws its power from the USB port of that computer, instead of a "powered" hub that draws its power from a mains electricity socket) then a '**failed to set power state**' error may occur if there is not enough power available from the computer to supply the power needs of both cameras at the same time.

As you are able to stream both cameras at the same time when using the hub with Ubuntu, and that you can view both streams in the RealSense Viewer on Xavier but not with the C++ script ... that would suggest that there may be something in the C++ script that causes the cameras to draw more power than the Viewer does, and that the Xavier board - unlike the Ubuntu machine - may not be able to cope with this power draw.

It looks as though your script is based upon the **rs-multicam** example's code in regard to how it sets up the pipelines.

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp

Does the **rs-multicam** example function with both cameras streaming on Xavier, please?

If your script is only using the pipeline setup section so far and not including the rest of the script, I wonder if the absence in that script of a control mechanism for incoming frame data (such as **poll_for_frames**) is causing the Xavier's CPU to be overwhelmed with 100% processor usage and resulting in the failure to set power state error.

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L44

Would it be possible to check your Xavier's CPU usage percentage at the moment when you run your script please and see if it is being maxed out at 100% usage?  Thanks!

longwoo, 2020-04-26:
@MartyG-RealSense Thank you for your answer. 
As I had updated my last reply.
The bug is caused by 
```
cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
pipe.start(cfg);
```

By the way,  before I comment the lines above, my Xavier is working on `MODE 30W ALL` with `jectson_clock` turned on, and my hub is an "powered" hub which draws its power from a mains electricity socket..

It works after I comment the lines mentioned above even my hub is an "unpowered" state.

MartyG-RealSense, 2020-04-26:
Have you tried commenting out your test line instead please and leaving the cfg lines uncommented:

 **std::cout<<"test1 ==== "<<std::endl;**

longwoo, 2020-04-26:
@MartyG-RealSense 

Yes, failed again when did so ,the same with **rs-multicam example**, test as follow
```
  
    // Start a streaming pipe per each connected device
    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        //uncomment this will rise a `failed to set power state`bug
        cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        // Start streaming with default recommended configuration
        pipe.start(cfg);
        pipelines.emplace_back(pipe);

    }
```

MartyG-RealSense, 2020-04-26:
The link below has an interesting discussion on the Nvidia forums from someone who had problems with multiple RealSense cameras on Xavier but not on a host macine.

https://forums.developer.nvidia.com/t/realsense-camera-unstable/71798

longwoo, 2020-04-26:
Yes, this link discussed about "USB SCP Overflow" error, which suffers a long time before Jectpack4.3 (r32.3) version.

This error not only occurs in multiple RealSense but also one RealSense, but with the help of JetsonHacks's patch, we can use it.

 Now  Jectpack4.3 (r32.3) , which I used, is stable and never shows this kind of error again. It is hard to tell there is some connection between "USB SCP Overflow" error and "failed to set power state" error.

MartyG-RealSense, 2020-04-26:
Can you confirm please that your program works if a single camera is attached.  That would help to demonstrate whether the problem was with the code or the number of cameras (the demand on the Xavier's resources), since the program should look for the serial number of the one camera that is attached.  If it works, that should show that the serial number retrieval code is working okay.

longwoo, 2020-04-26:
Your guess is right. It works if a single camera is attached. And crash for two. 

```
    rs2::context ctx;// Create librealsense context for managing devices
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    std::vector<rs2::pipeline> pipelines;

    //Add desired streams to configuration
    const int RGB_STREAM_WIDTH = 640;//1280;
    const int RGB_STREAM_HEIGHT = 480;//720;
    const int FRAME_RATE = 30;
    const int DEPTH_STREAM_WIDTH = 848;
    const int DEPTH_STREAM_HEIGHT = 480;

    // Start a streaming pipe per each connected device
    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        //uncomment this will rise a `failed to set power state`bug
        cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        std::cout<<"serial num = "<<dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER)<<std::endl;

        cfg.enable_stream(RS2_STREAM_COLOR, RGB_STREAM_WIDTH, RGB_STREAM_HEIGHT, RS2_FORMAT_BGR8, FRAME_RATE);
        cfg.enable_stream(RS2_STREAM_DEPTH, DEPTH_STREAM_WIDTH, DEPTH_STREAM_HEIGHT, RS2_FORMAT_Z16, FRAME_RATE);
        pipe.start(cfg);
        pipelines.emplace_back(pipe);

    }

    std::cout<<"start"<<std::endl;
```

MartyG-RealSense, 2020-04-26:
You could try changing the frame rate value at the top of your script from '30' to '15' to see whether the power state failure is occurring because 2 x two streams (RGB and depth) at 30 FPS is too much for the board's available resources when 2 cameras are attached.

longwoo, 2020-04-27:
I share the same point as @tispratik , `I do not think it is a power issue at all`.

First,  it works as I did the experiment before,
```
    rs2::context ctx;// Create librealsense context for managing devices
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    std::vector<rs2::pipeline> pipelines;

    rs2::align align_to_color(RS2_STREAM_COLOR);

    //Add desired streams to configuration
    const int RGB_STREAM_WIDTH = 640;//1280;
    const int RGB_STREAM_HEIGHT = 480;//720;
    const int FRAME_RATE = 30;
    const int DEPTH_STREAM_WIDTH = 848;
    const int DEPTH_STREAM_HEIGHT = 480;

    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        //uncomment this will rise a `failed to set power state`bug
        //cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        std::cout<<"serial num = "<<dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER)<<std::endl;

        cfg.enable_stream(RS2_STREAM_COLOR, RGB_STREAM_WIDTH, RGB_STREAM_HEIGHT, RS2_FORMAT_BGR8, FRAME_RATE);
        cfg.enable_stream(RS2_STREAM_DEPTH, DEPTH_STREAM_WIDTH, DEPTH_STREAM_HEIGHT, RS2_FORMAT_Z16, FRAME_RATE);

        // Start streaming with default recommended configuration
        pipe.start(cfg);
        pipelines.emplace_back(pipe);

    }

    while (true)
    {

        std::vector<rs2::frameset> vfs;
        try {
            for (auto &&pipe : pipelines)
            {
                rs2::frameset fs_t;
                if (pipe.poll_for_frames(&fs_t))
                {
                    vfs.emplace_back(fs_t);
                }
            }

            // Convert the newly-arrived frames to render-friendly format
            for (const auto& fs : vfs)
            {
                // Get the serial number of the current frame's device
                auto serial = rs2::sensor_from_frame(fs)->get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);

                auto align2c_frames = align_to_color.process(fs);

                auto align2c_color = align2c_frames.get_color_frame();
                auto align2c_depth = align2c_frames.get_depth_frame();

                cv::Mat align2c_Color(cv::Size(align2c_color.get_width(), align2c_color.get_height()), CV_8UC3, (void*)align2c_color.get_data(), cv::Mat::AUTO_STEP);

                if( std::string(serial) == "83XXXXXXXXXX"){//D435i
                    cv::imshow("test1", align2c_Color);
                    cv::waitKey(1);
                }
                else if( std::string(serial) == "94XXXXXXXXXX"){//D415 
                    cv::imshow("test2", align2c_Color);
                    cv::waitKey(1);
                }
            }

        }

        catch (const rs2::error & e)
        {
            std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
            continue;
        }
    }
```

Second ,  it failed as I used the framerate `15` with `cfg.enable_device` as you recommend.

MartyG-RealSense, 2020-04-27:
At this point, it would be worth trying a "bypass" installation method for Librealsense if you have not done so already.  It should work with any type of Linux and is not dependent on kernel versions or patches, so should help to eliminate the possibility that the problem is being caused by the Linux kernel.  An internet connection is required for this method of installation.

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

longwoo, 2020-04-27:
As far as I know, this method was mentioned in this topic before. In fact, I had already read this and  tried it when I first met this problem.  Unfortunately, it failed.

MartyG-RealSense, 2020-04-27:
Oh yes, sorry.  I did check for previous mention of it using the search keyword 'bypass' (this discussion is difficult to search manually due to its length!).  I had mentioned it before but not in terms of 'bypass'.  I apologise for that.  Let's do a summary:

@tispratik   Your problem was that when connecting six cameras to an i7-9800X computer, the 6th camera  could not use the motion frame.

Some of the errors that you reported in your initial test, such as '**Permission Denied**' and '**Entity type for entity Processing was not initialized!**' suggest that the **udev rules** in your installation may need updating.  You can do this by running the command below from the Librealsense root directory:

**sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ sudo udevadm control --reload-rules && udevadm trigger**

tispratik, 2020-04-27:
@MartyG-RealSense my problem is when connecting two cameras. I am frustrated that you are pushing the ball around and suggesting workarounds and things to try when enough users have reported a problem. This should be look into in detail by Intel engineers and come up with a solution/patch soon. We cannot go around this way.

[30 more replies]

closed 2020-06-16 by RealSenseSupport