realsenseai/librealsense #5829: T265- getting pose and fisheye frame

galikrosentalik, 2020-02-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  0.2.0.926 | 
| Operating System & Version |  Linux (Ubuntu 18)  | 
| Kernel Version |  4.15.0-54-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.32.1  | 
| Language                            | C++ | 
| Segment			|  Robot | 

### Issue Description
Hey all, 
my system is multi cam system, same process, but different threads is handling both D435/415 and T265.
The system needs to work with more then one of a kind, currently 2 * T265 and 1 * D4X5.
When getting from the T265 only positioning, the system works fine.
We need to expend our system so the T265 will generate in addition to pose data also the fish-eye frames.
Our current flow does not use callback methods, but the pipe.poll_for_frames.
When enabling also the fish-eye stream we are not able to keep with the same needed FPS for the positioning, the FPS got lower to 30.
I tried to change the flow to work like your example pos-and-image. when working with single T265 camera everything works fine, but when trying to integrate it in multi-cam system the callback just stooped been called.
are you familiar with the issue of integrating  multi-cam and pose-and-image?

galikrosentalik, 2020-02-19:
???

RealSenseCustomerSupport, 2020-02-22:
----------------------------------------------

Hi @galikrosentalik

We understand there are issues with multi cam scenario.

Read through this thread:  https://github.com/IntelRealSense/librealsense/issues/5614

Also a PR that can also help:  https://github.com/IntelRealSense/librealsense/pull/5645

galikrosentalik, 2020-03-09:
I read the other threads comment.
I did not understand how this issue is related to mine, do you think the last pull request approved by dorodnic suppose to solve my as well?

closed 2021-07-07 by RealSenseCustomerSupport