realsenseai/librealsense #5830: Unable to debug T265, still..

galikrosentalik, 2020-02-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |  Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |4.18.0-25| 
| Platform                                 | PC |
| SDK Version                            | 2.24.0| 
| Language                            |  C++ | 
| Segment			|  Robot | 

### Issue Description

Hi, sorry for the delay,I still suffering from the issue #4500, can we re-open the issue?
is there any more info you need from me in order to investigate this issue?

zainmehdi, 2020-02-17:
I am having the same issue with T265 using qtcreator as an IDE. @RealSenseCustomerSupport  Please provide some hints on how to fix it.

galikrosentalik, 2020-02-19:
???

galikrosentalik, 2020-03-09:
???????

RealSenseSupport, 2020-09-11:
Thank you for highlighting the continued issue of putting system in a debug environment while using T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-17 by RealSenseSupport