realsenseai/librealsense #5831: Building SDK Samples for Android OS

Colinsiya, 2020-02-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Host machine | Linux with Ubuntu 18.04 | 
| Platform                                 | Rooted Android target device |
| CMake version | 3.10.2 |
| ANDROID_ABI | armeabi-v7a |
| SDK version | 2.0 |
| NDK version | Android ndk r21 |

### Issue Description

I built the SDK samples for Android OS following all the steps described in this document (page 26) : https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-SDK2-Github-Guide.pdf

I run the executable file: 
`./rs-depth`
And I get the following error:
**`/system/bin/sh: <stdin>[2]: ./rs-depth: not executable: 64-bit ELF file`**

For information: 

- `$ file rs-depth`

`rs-depth: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/l, for GNU/Linux 3.2.0, BuildID[sha1]=3bf14a7cdd2239fdc40cae00909a57adedc62f9a, with debug_info, not stripped
`

- ` $ file librealsense.so`

`librealsense2.so.2.32.1: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=ee9cb98573f48590f464f905113bf70e077b04b2, with debug_info, not stripped`

On the Android device through adb:

- `$ uname -a`
`Linux localhost 4.9.112+ #1 SMP PREEMPT Fri Mar 15 09:53:25 PDT 2019 aarch64`

I tried to build this samples with other ABIs but I always have the same error.

So I can't execute the camera's samples programs on my target device.

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@Colinsiya  From the information you provided, you're using armeabi-v7a. What did you configure for "ANDROID_ABI" when build?

cmake .. -DANDROID_ABI= - DCMAKE_TOOLCHAIN_FILE=/build/cmake/android.toolchain.cmake

Colinsiya, 2020-02-13:
@dorodnic When I build the librealsense I use the parameter "armeabi-v7a" for "ANDROID_ABI".

`$ cmake .. -DANDROID_ABI=armeabi-v7a -DCMAKE_TOOLCHAIN_FILE=/build/cmake/android.toolchain.cmake`

I am using a phone with android 5.1.1, with an ARMv7 Processor rev 3 (v71).

Hope this is the answer to your question, if not, let me know, and I'll give you more information. Thanks a lot !

RealSenseCustomerSupport, 2020-02-18:
----------------------------------------------

@Colinsiya Could you please check the environment variable setting of ANDROID_NDK?

RealSenseCustomerSupport, 2020-03-11:
----------------------------------------------

@Colinsiya Did you get it through? Could you please update? Thanks!

Colinsiya, 2020-03-12:
Sorry for the delay ! Unfortunately, I could not invest more time on this subject. Thanks a lot !

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@Colinsiya Thanks for the reply! Let's close it at this point. Please create new one if you still have issues or questions. Thanks!

closed 2020-03-17 by RealSenseCustomerSupport