realsenseai/librealsense #5832: Question: Choosing specs to Intel D435

pascoa-pand, 2020-02-11:
| Camera Model                       | { **D400** } | 
| Platform                                 | {**PC**} |
| Language                            |  {**opencv**/**python** }                          | 
| Segment			|  {**others** }                          | 

Hello all,

I've recently acquired a RealSense camera - Intel D435. I want to take full advantage of the camera, getting the most FPS and highest resolution possible with depth and rgb. In my use case, I am using depth to remove background from the rgb frames.

I am about to buy a new PC from where I will run the Intel camera. 
What specs should I be looking for to get a good performance with the least amount of money? Is the processing happening more on CPU or GPU or both? Could you recommend a range of CPUs or GPUs that could provide an excellent experience with this camera?

Thank you for your support.

kafan1986, 2020-02-11:
Maybe for your case that you described D415 was a better choice, as the D415's FOV of both colour and depth match, whereas D435 depth sensor has much wider FOV compared to the RGB camera's FOV.

Anyways coming back to your question, the RGB camera produces images at 30 FPS and depth is data can be obtained upto 90 FPS (at different resolution), but for your use case you can obtain highest resolution depth data at 30 FPS. Both the depth and colour image are processed on the D435 using the internal Vision D4 board, so the heavy processing part has already been done for you. Some amount of CPU might be used to bring the raw data from USB to CPU.

But for your particular use case you might face problem as you will be required to align the colour and depth image. A high clocked CPU will probably give you around 10 to 15 FPS of aligned data. If you can implement a suitable multi-threaded solution, you should probably be able to process it on the CPU alone, given all modern processors have 4 or above cores available. May be if you face any problem you can add a mid-range NVIDIA gpu later on as librealsense has a CUDA optimized alignment functionality available.

Basically get a computer with mid-range CPU and you should be good. Also, make sure you have USB 3.0 or above port available (although it should be, if it is manufactured in last 5 years)

MartyG-RealSense, 2020-02-11:
Thanks so much @kafan1986 

If you anticipate using the camera for processing-intensive applications, I would recommend going for an i7 processor.

As Kafan said, processing will happen on the CPU by default since the Vision Processor D4 component inside the camera will do initial processing that a computer GPU would normally do if the D4 was not there.

If you get a machine with Nvidia graphics then you can build Librealsense with CUDA support to offload work onto the GPU, as Kafan also said.  For non-Nvidia graphics, you can alternatively use something called a GLSL Processing Block to offload to the GPU.  Details are in the link below.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

If you wish to save money by choosing an older Intel processor generation, I would recommend a PC with a 7th generation Kaby Lake processor as a minimum, i7 if possible.

pascoa-pand, 2020-02-11:
This was very enlightening. Thank you very much for your help and kind responses

closed 2020-02-11 by pascoa-pand