realsenseai/librealsense #5833: when using function "points.export_to_ply()",the Z value of the derived point cloud is negative

DreamLFairy, 2020-02-12:
After I used the function of "points.export_to_ply()",I used the software "MeshLab" to view the point cloud,I found that the Z value of the derived point cloud is negative .
You know,the object is in front of the camera, so the distance is supposed to be a postive number.
Thank you  for answering my question very much.

ev-mp, 2020-02-12:
@DreamLFairy , that's correct.
During PLY export conversion the pointcloud coordinate system (CS) is re-aligned from East-Down-North (x-right,y-down,z-forward) to East-Up-South (x-right, y,-up, negative z - forward) for native compatibility with OpenGL CS and tools that  adhere to it such as MeshLab.
Note that during the transformation the  CS's "Right-handedness" is preserved.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Closing this ticket. Please feel free to let us know if there is any other question.
Thank you.

closed 2020-03-03 by RealSenseCustomerSupport