realsenseai/librealsense #5835: using new hardware_reset api causing application to stuck

galikrosentalik, 2020-02-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 & D400 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version | Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  4.15.0-54-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.32.1 | 
| Language                            | C++ | 
| Segment			| Robot | 

### Issue Description
Hey all, 
my system is using your new api extension including the hardware_reset method.
my system is multi cam system, same process, but different threads are handling both D435/415 and T265.
The system needs to work with more then one of a kind, currently 2 * T265 and 1 * D4X5.
We are now testing the very much needed hardware_reset method.
When working with single device everything seems to work pretty well, we noticed some problems when working with more then one device.
In case the hardware_reset is been called while other thread is trying to call pipe.start() or in case  that more then one thread is calling the hardware_reset simultaneously, we notice our application is getting segmentation fault.
are you familiar with the issue?

galikrosentalik, 2020-02-19:
???

RealSenseCustomerSupport, 2020-02-22:
----------------------------------------------

Hi @galikrosentalik

Similar to your other github thread.

Read through this thread:  https://github.com/IntelRealSense/librealsense/issues/5614

Also a PR that can also help:  https://github.com/IntelRealSense/librealsense/pull/5645

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @galikrosentalik

Similar to your other github thread.

Read through this thread:  https://github.com/IntelRealSense/librealsense/issues/5614

Also a PR that can also help:  https://github.com/IntelRealSense/librealsense/pull/5645

Let us know if this helps.

***If we don’t hear from you in 7 days, this issue will be closed.**

galikrosentalik, 2020-03-09:
I read the other threads comment.
I did not understand how this issue is related to mine, do you think the last pull request approved by dorodnic suppose to solve my as well?

RealSenseSupport, 2020-09-11:
Thank you for highlighting this issue with hardware_reset while using T265 along with other RealSense products. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue with the T265.

closed 2020-09-17 by RealSenseSupport