realsenseai/librealsense #5839: How to make sure the same frame is taken when two cameras are turned on?

DreamLFairy, 2020-02-13:
The object I want to shoot is moved on a conveyor belt, and at the same time there are many different objects.I need two cameras to shoot an object at the same time, and then .So I want my two cameras to take the same frame.How can this be done?Thank you very much for your answers,I will look at them one by one.By the way,how can I reply to your reply?

MartyG-RealSense, 2020-02-13:
To reply to a comment, you can write a message in the text box at the bottom of the page and click the green 'Comment' button.

I recall a case where someone aimed to trigger a RealSense camera on a conveyor belt when objects pass by a photoelectric switch.  When an object passes through a light beam and breaks it, an electrical pulse (such as one used to trigger a camera) can be generated.

Another approach is using an external flash as a camera sync trigger.  An example of this is in the link below.

https://github.com/IntelRealSense/librealsense/issues/4574

In regard to observing many objects on a conveyor with a camera, you may find useful a project where a camera scanned and identified multiple Lego bricks as they moved quickly under the camera.

https://github.com/IntelRealSense/librealsense/issues/5393

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

Hi DreamLFairy

0) Sounds you're looking for the multi-cam sync feature (https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration)
1) What kind of RealSense cameras are you using now? two D435 or two D415? (there is a limitation on this feature for D435)
2) Then what camera sensors are you looking for "to shoot object at same time"? Depth vs Depth, or Depth vs RGB?
3) To verify the sync feature, you can use a high resolution Stop Watch. See attached picture.

Please let us know your setup, so we can provide more info for your case.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

Attachment(s):
D435_Depth_RGB_sync.JPG - https://realsensesupport.intel.com/attachments/token/NULF5eidSLQ5crqOwgy6sRB71/?name=D435_Depth_RGB_sync.JPG

closed 2020-02-26 by RealSenseCustomerSupport