realsenseai/librealsense #5841: How to set point cloud not to fit

DreamLFairy, 2020-02-13:
I hung the object on a trolley, it was at a distance from the trolley.But the generated point cloud and the cart are connected.This makes it difficult for me to filter out the backgroud.I hope it doesn't fit points that don't exist in reality.I'm quiet keen on your answering. Thank you for answering very much;

MartyG-RealSense, 2020-02-13:
If the trolley and the object are at the same distance (the object is hanging from the trolley) then you probably could not cut the trolley out by filtering out the background beyond a certain depth.  A better method may be to define a 'bounding box' around the area where the object is, so that only the points inside the box are used and everything outside of the box is ignored.

The Lego brick conveyor system article I linked to in your post about conveyor capture has some introductory information about use of bounding boxes.

DreamLFairy, 2020-02-13:
First, how do I upload pictures?

I think you got me wrong. I use another brand of camera, which can be separated at different distances, but with the intel d345, the same conditions and objects cannot be separated by distance. I found that the generated point clouds are automatically fitted, so that point clouds of different distances are connected together. Thank you for you reply.

MartyG-RealSense, 2020-02-13:
You can add pictures to the message by dragging an image file into the comment box, or copying the image in a paint program and Pasting it into the comment window

DreamLFairy, 2020-02-13:
I cann't upload a picture.It said"Something went really wrong,and we can't proces the file,Try again".
But I tried many times and it still has errors.
Is there any other way to communicate?

MartyG-RealSense, 2020-02-13:
you could try posting your image at the other RealSense forum, where I also answer messages.

https://support.intelrealsense.com/hc/en-us/community/topics

With RealSense you can define a 'depth clamp' with a minimum distance and a maximum distance, so that values outside of that range are ignored. An easy way to test this is in the **Depth Table** sub-section of the RealSense Viewer's **Advanced Mode** side-panel controls.

[image]

MartyG-RealSense, 2020-02-13:
I will post your images here so that the RealSense forum team members can see them easily too.

"The first picture is Intel's camera,the second picture is other's camera.The third picture is the difference;  This makes it difficult for me to filter out the background.  I hope it doesn't fit points that don't exist in reality".

[image]

[image]

[image]

MartyG-RealSense, 2020-02-13:
In the first image taken with the RealSense camera, it looks as though the surface between the trolley and object is actually there (e.g the top of the trolley).  It might be difficult to remove if that is the case.

One way to get the RealSense camera to ignore that area might be to put a black surface on top of the trolley, as depth cameras have difficulty seeing black colored areas and may render it on the image as an empty black area because it cannot read depth detail in that area of the scene.

DreamLFairy, 2020-02-14:
Thank you for your reply very much. Is it possible to remove the horizontal point clouds connected in the picture?The two vertical boards are separated, but they are not actually connected.The effect of the second picture is consistent with reality. I also hope that Intel's cameras can achieve this situation.

MartyG-RealSense, 2020-02-14:
Looking very carefully at the image made by RealSense, the depth of the horizontal strip looks slightly behind the upper block.  Is there a wall behind the object that the camera might be seeing through the gap in between object and trolley?  If there is, simply setting a Max Depth clamp should remove the wall from the image.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Closing this issue if there is no other question.
Thank you.

closed 2020-03-03 by RealSenseCustomerSupport