realsenseai/librealsense #5843: Python example for T265 relocalization event notification

hamsterasu, 2020-02-13:
Hi, can anyone show me an Python example for setting up a relocalization event notification for T265. I have referred to this example https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-advanced but this is in C++.

MartyG-RealSense, 2020-02-13:
I hope that the link below will be helpful to you.  It makes reference to a problem with firmware at the time that the comment was written (March 2019) but they released an updated firmware later.

https://github.com/IntelRealSense/librealsense/issues/3516#issuecomment-474828004

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

Also want to add that there are no plans on adding any further examples for T265

Here is the T265.md:  https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/doc/t265.md
Outlines the examples available for T265

Let us know if anything further is needed

hamsterasu, 2020-02-18:
Sorry for late reply, I got it working. Thanks for the links. I got a working Python code snippet in Python for anyone who may need it:
```
def callback_function(notif):
    if notif.get_category() is rs.notification_category.pose_relocalization:
        print("Relocalization has happened!")

pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.pose)
device = cfg.resolve(pipe).get_device()
pose_sensor = device.first_pose_sensor()
pose_sensor.set_notifications_callback(callback_function)
pipe.start(cfg)
```

closed 2020-02-18 by hamsterasu

MartyG-RealSense, 2020-02-18:
Thanks for sharing your code with the community!   Glad you found a solution  :)