realsenseai/librealsense #5845: Intel Realsense SR300 with Jetson Nano produces incorrect depth values in Ubuntu

kevinpaulsudarshan, 2020-02-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |   4.9.14       | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            | 2.0                        | 
| Language                            |  C++ | 
| Segment			|  Computer Vision | 

### Issue Description
Hi, I am new to the realsense SDK. I am trying to work on the depth values obtained from the realsense in c++ on Jetson nano. I noticed that the depth values obtained in the realsense-viewer does not reflect the real distance to the object. The max value I get is 0.13m. But when I connect the same camera to my desktop which is running Windows 10, I can see that the depth values obtained are very close the actual distance to the object. Can you please tell me if there is something wrong i am doing. For reference see below the snapshots from realsense-viewer. Thank you in advance for your help and answers.

Windows:
[image]

Ubuntu(Jetson nano)
[image]

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@kevinpaulsudarshan Seems the snapshot of **Ubuntu(Jetson nano)** doesn't show depth. Could you please provide another snapshot and specify where you saw 0.13m difference?

kevinpaulsudarshan, 2020-02-14:
Here are two snapshots of the same scene on Windows and Ubuntu respectively. In **Ubuntu I have set the "Motion to depth range" to 220. In Windows the same parameter is 9**. The distance to the object on Ubuntu is 0.12m and on Windows is 0.56m(left bottom corner of the image in realsense-viewer). The distance to object on Ubuntu always caps out at 0.13m. **0.13m is not the difference but the maximum distance shown when I move the cursor over the image in realsense-viewer**.

Windows: 
[image]

Ubuntu:
[image]

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

@kevinpaulsudarshan Did you get such issue if you set **"Motion to depth range"** to 9 on Ubuntu? And could you please provide SDK version you used? Thanks!

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@kevinpaulsudarshan I couldn't reproduce the issue with the latest SDK. Could you please try with the latest SDK also? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@kevinpaulsudarshan Any update from your side?Thanks!

RealSenseCustomerSupport, 2020-04-01:
----------------------------------------------

@kevinpaulsudarshan Sorry that we didn't get response from your side for weeks. Close it at this point. Please feel free to create another one if you still have questions or issues. Thanks!

closed 2020-04-01 by RealSenseCustomerSupport