realsenseai/librealsense #5846: SR305 on TX2(Jetpack4.2) Realsense-viewer get depth value error

weseu, 2020-02-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | 2.31.0 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 | NVIDIA TX2 Jetpack4.2  |

### Issue Description
when running realsense-viewer I get the error depth.
[image]

I try the following  way and get the following result.
1.  when I down librealsense  version to 2.22, I get the correct depth value. 
[image]
but I want to use it on Ros. so I use the realsense2_camera package.  and than I get the following massage.
[image]

2.  when I update librealsense version to 2.24. I get the incorrect depth value. 

3.  when I update librealsense version to 2.31. I get the incorrect depth value,too. 

by the way.........
1. If I use D435i, I can get currect depth value on TX2 ROS and realsense-viewer.
2. I can use D435i and SR305 on PC getting correct depth value,too.

please tell me how to get the correct depth.

RealSenseCustomerSupport, 2020-02-15:
----------------------------------------------

Is there any difference (2.22 vs 2.31) in the "depth units" under the Control tab?

weseu, 2020-02-15:
how do I know  the "depth units" under the Control tab?

weseu, 2020-02-15:
it show 0.000125.

[image]

weseu, 2020-02-15:
By the way. If I use version 2.22.0, I can get currect depth value 2cm to 1.5m.
but I use version 2.24 or later. I will get the following relevance.

actual distance          depth distance
5 cm                                  0.05m
10cm                                 0.10m
15cm                                 0.07m
20cm                                 0.06m
30cm                                 0.04m

weseu, 2020-02-17:
By the way. when I use realsense_ros 2.2.6 and librealsense 2.22.0, and folow https://github.com/IntelRealSense/realsense-ros/pull/844/commits/150ecb9bc47a6b1f6ee99e166d9e8b589aec8711`` to fix the code.

running
roslaunch realsense2_camera rs_camera align_depth:=true

and running
realsense-ros/realsense2_camera/scripts/show_center_depth.py
 
I get the following depth value, it is like the realsense-viewer with version 2.24 or later.

| actual distance   | depth distance      |
|------------------|--------------- |
| 0.05m  | 0.05m | 
| 0.1m | 0.1m | 
| 0.15m |   0.07m | 
| 0.2m | 0.06m  | 
| 0.3m    | 0.04m  |
| 0.5m~    | 0.02m  |

please help me.

RealSenseCustomerSupport, 2020-02-20:
----------------------------------------------

What is the camera firmware version. The ticket info shows the firmware is 2.31.0 but it looks like the RealSense SDK build version.
Anyway, I just ran a quick test on SR305 (v.3.27.3) with RealSense Viewer (v.2.32.1) manually built on TX2 and the depth reading works as expected.
Thank you.

weseu, 2020-02-20:
My camera SR305 firmware version is 3.27.3.0.
and can I try your TX2 jetpack version??  maybe if I use your TX2 jetpack version,Realsense viewer version. it will work.  (I never used realsense viewer 2.32.1 )

so, can you tell me what version is your jetpack.

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

The Jetson TX2 has the Jetpack 4.2.2 - L4T 32.2.1 Kernel 4.9.140. (see below).

    $ cmake ../ -DBUILD_EXAMPLES=true -DFORCE_RSUSB_BACKEND=true
    $ head -n 1 /etc/nv_tegra_release
    # R32 (release), REVISION: 2.1, GCID: 16294929, BOARD: t186ref, EABI: aarch64, DATE: Tue Aug 13 04:45:36 UTC 2019
    $ uname -r
    4.9.140-tegra

    $ ./examples/C/distance/rs-distance
    There are 1 connected RealSense devices.

    Using device 0, an Intel RealSense SR300
        Serial number: 636203005000
        Firmware version: 3.27.3.0

    The camera is facing an object 0.983 meters away.
    The camera is facing an object 0.942 meters away.
    The camera is facing an object 0.897 meters away.
    The camera is facing an object 0.851 meters away.
    The camera is facing an object 0.810 meters away.
    The camera is facing an object 0.774 meters away.
    The camera is facing an object 0.741 meters away.
    The camera is facing an object 0.705 meters away.
    The camera is facing an object 0.671 meters away.
    The camera is facing an object 0.637 meters away.
    The camera is facing an object 0.604 meters away.
    The camera is facing an object 0.574 meters away.
    The camera is facing an object 0.542 meters away.
    The camera is facing an object 0.509 meters away.
    The camera is facing an object 0.482 meters away.

weseu, 2020-02-25:
I try to update TX2 jetpack 4.2.2 and librealsense 2.32.1 , and I get the same issue......!!
The following is my installation method.
1. git clone github https://github.com/jetsonhacks/installRealSenseSDK ,
and fix the buildLibrealsense.sh for installing librealsense 2.32.1
[image]

2. runing installLibrealsense.sh and buildLibrealsense.sh.

3. running realsense-viewer..and I get the following result
[image]
[image]

ps. the terminal show
[image]

4. running  roslaunch realsense2_camera rs_camera and show_center_depth.py, I get the following result
[image]

please help me, or give me some direction.....

weseu, 2020-02-25:
by the way. If I running ./examples/C/distance/rs-distance, I get the following result.
[image]

drfg0119, 2020-02-27:
Hi @RealSenseCustomerSupport and @weseu , I got the similar problem too [https://github.com/IntelRealSense/realsense-ros/issues/1087](url)
I am really frustrated about this...

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

It may be related to cuda optimization of depth conversation for SR300 on Jetson TX2.
Please build with -DBUILD_WITH_CUDA=false and check if the depth value return as expected.
Thank you.

weseu, 2020-03-09:
$ cd librealsense/build
$ cmake ../ ../ -DBUILD_EXAMPLES=true -DFORCE_LIBUVC=true -DBUILD_WITH_CUDA=false -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=bool:true -DFORCE_RSUSB_BACKEND=true
$ make -j4
$ sudo make install
$ realsense-viewer

and then,I get the same issue.......

weseu, 2020-03-10:
or........did you have any method can let me follow for installing my TX2 enviroment.

because this issue let me be very worry...

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

@weseu, here is bash history.

      76  2020-03-10 11:03:33 mkdir build
      77  2020-03-10 11:03:38 cd build
      78  2020-03-10 11:04:02 cmake ../ -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=false -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=bool:true -DFORCE_RSUSB_BACKEND=true
      79  2020-03-10 11:04:19 make -j4
      80  2020-03-10 11:54:55 ./tools/realsense-viewer/realsense-viewer
      81  2020-03-10 11:56:03 ./examples/C/distance/rs-distance

The device points at a flat wall which is 0.5m away and the rs-distance returns expected value

    nvidia@nvidia-desktop:~/librealsense/build$ ./examples/C/distance/rs-distance
    There are 1 connected RealSense devices.

    Using device 0, an Intel RealSense SR300
        Serial number: 636203005000
        Firmware version: 3.27.3.0

    The camera is facing an object 0.500 meters away.
    The camera is facing an object 0.500 meters away.
    The camera is facing an object 0.498 meters away.
    The camera is facing an object 0.498 meters away.
    The camera is facing an object 0.498 meters away.
    The camera is facing an object 0.499 meters away.
    The camera is facing an object 0.499 meters away.

And here is the screenshot of the realsense viewer also for reference:
https://intel.sendsafely.com/receive/?thread=5YAM-X7DE&packageCode=qz3asxShOzmBVG6PBNaa2nZtnNZyrTZDrC5oUTmg8HI#keyCode=bz2vmDe5TOHDhg9pBhNOLtl6oEEKqgxnjYkd4-NdD4c

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

Let me re-attach the image

Attachment(s):
sr305_lrs2.32.1_fw3.27.3.png - https://realsensesupport.intel.com/attachments/token/RNg5JBdkJOTIvMtfCGIDt5UQ1/?name=sr305_lrs2.32.1_fw3.27.3.png

weseu, 2020-03-13:
wow~~  I success to get the expected value.
but it let me confuse when I key in "sudo make install",and key in "realsense-viewer". I will get the same issue.  so~~why /usr/local/bin/realsense-view can't work!!!

by the way, I test the new realsense_ros (2.2.13),it can work!!!

thank you very much~~~

RealSenseCustomerSupport, 2020-03-13:
----------------------------------------------

@weseu, you are welcome and glad to know that it works now.
If you have installed from the deb package, please purge it first before running "sudo make install" again.
Thank you.

weseu, 2020-03-16:
I find a issue,but it didn't affect me.

$ roslaunch realsense2_camera rs_camera 
[image]

RealSenseCustomerSupport, 2020-03-16:
----------------------------------------------

Thank you for the update @weseu.
For the new issue, please give udev installation script a try, as suggested in the #5820.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
Please feel free to open a new issue if problem persists.
Thank you.

weseu, 2020-03-18:
thank you. I close the issue first.  If  I get the new issue,I will open a new issue.

closed 2020-03-18 by weseu