realsenseai/librealsense #5847: Manual frameset creation

jc211, 2020-02-14:
Hi,
I have one depth frame and one color frame. I want to create a frameset myself of these two. Is this possible?

MartyG-RealSense, 2020-02-14:
There was a previous case in 2019 where someone tried to create framesets manually using **software device**.  Their work may provide a useful reference for you.

https://support.intelrealsense.com/hc/en-us/community/posts/360037090133-Construct-Frameset-Object

dorodnic, 2020-02-16:
Hi @jc211 @MartyG-RealSense 
Another way to do this is using custom processing block:
```cpp
std::vector<rs2::frame> bundle;
rs2::processing_block bundler([&](rs2::frame f, rs2::frame_source& src) {
    bundle.push_back(f);
    if (bundle.size() == 2)
    {
        auto fs = src.allocate_composite_frame(bundle);
        src.frame_ready(fs);
        bundle.clear();
     }
});
rs2::frame_queue q;
bundler.start(q);
```
Use like this:
```cpp
bundler.invoke(f.get_depth_frame());
bundler.invoke(f.get_color_frame());
rs2::frameset fs = q.wait_for_frame();
```
In either of the methods you need some object to own any frames you want to create (either software device or processing block). This is because all frames in the SDK are recycled for performance reasons, so they must be part of some larger class.

jc211, 2020-02-16:
@dorodnic Thanks, that works!

closed 2020-02-16 by jc211

DonLuyendijk, 2021-11-12:
Hi,

I would also like to use the custom processing block example in Python. I tried to translate it, but I can't get it to work.
I made a script that should create a frameset from a color frame and depth frame. The script crashes at the `q.wait_for_frame()` call, with the message `RuntimeError: Frame did not arrive in time!`. I then added a print call to the bundler callback, but nothing is printed. It seems like the callback is never called when `bundler.invoke()` is called.
Can i get some help with this issue?

Here's the script:
```py
import pyrealsense2 as rs

def bundler_callback(f, src):
	print('bundler function called')
	bundle.append(f)
	if len(bundle) == 2:
		fs = src.allocate_composite_frame(bundle)
		src.frame_ready(fs)
		bundle.clear()

pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file('stairs.bag')
pipe.start(cfg)

fs = pipe.wait_for_frames()

bundle = list()
bundler = rs.processing_block(bundler_callback)
q = rs.frame_queue()
bundler.start(q)

bundler.invoke(fs.get_color_frame())
bundler.invoke(fs.get_depth_frame())

fs = q.wait_for_frame()
```

And here's a link to the sample file I used:
https://librealsense.intel.com/rs-tests/TestData/stairs.bag

MartyG-RealSense, 2021-11-12:
Hi @DonLuyendijk  There are not any references about custom processing blocks in Python that I can point you to.  Looking at your script though, I would suspect that you may need to place the name of your callback within the pipe start instruction's brackets.  A Python example of this is demonstrated in a script at https://github.com/IntelRealSense/librealsense/issues/5417

For example: **pipe.start(cfg, bundler_callback)**