realsenseai/librealsense #5848: Remove the first X frames from a recorded bagfile

g2-bernotas, 2020-02-14:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |   Win (8.1/10)  |                                        | 
| Platform                                 | PC  |
| Language                            |  C++/python                      | 

### Issue Description

I have recorded bagfiles using my developed solution, which is similar to the one provided in the [RS-Record-Playback](https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp) example, except I used to pause/resume the recording ([#4633](https://github.com/IntelRealSense/librealsense/issues/4633)). 

I noticed that while the camera was being set-up as a recorder before it is being paused (the following line), it would have already started to record (0-3 seconds). As I **do not want** this information for my processing, I was wondering how can I remove these frames. I have looked around the Internet and Github repository for splitting/cropping the bagfiles files ([ros_bag_splitter.py](https://github.com/vnaveen9296/bag_splitter/blob/master/ros_bag_splitter.py), [cut.py](http://wiki.ros.org/bag_tools#cut.py), [ROS filter](http://wiki.ros.org/rosbag/Commandline#filter), [#1868](https://github.com/IntelRealSense/librealsense/issues/1868)); however, I couldn't find an easy solution, especially the one that does not involve working with ROS ([rs-convert tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert) provides frames only, but not a new, shorter, bagfile). 

I can skip a predefined number of frames during my processing; however, the number of unusable frames varies from file to file (I have over 1,000 files where the front padding ranges from 0 to 3 seconds, so from 0 to 90 frames at 30 FPS meaning that a single suits-it-all number does not work (I could choose the biggest number (90), but then I would lose a lot of data for the one that has no front padding). Another option would be to create a list of how many unnecessary frames each file has at the beginning; however, this could easily get lost, so I would rather have shorter bagfiles instead. 

My preferred language is Python, but it seems libraries such as [pyrosbag](https://pypi.org/project/pyrosbag/) are incomplete. 

TL;DR: how to remove the first X frames from a bagfile, but have it as a shorter bagfile (not individual frames).

MartyG-RealSense, 2020-02-14:
ROS has a toolkit called bag_tools that features a program called cut.py.  It can input a bag file, crop it to the specified start point and 'duration' end-point, and write the result as an output bag.

http://wiki.ros.org/bag_tools#cut.py

MartyG-RealSense, 2020-02-14:
You covered every bag editing possibility that my own research could turn up.  If your aim is to prevent the camera recording for the first few seconds after the camera starts, the easiest approach would likely be to simply pause the recording for a few seconds and then un-pause it.

Doing so with Python is discussed in the link below in a case where another user wanted to avoid recording the first couple of seconds whilst the camera settled down.

https://github.com/IntelRealSense/librealsense/issues/4249

g2-bernotas, 2020-02-14:
@MartyG-RealSense thank you for your response. I will use the cut.py then or skip X frames. I was wondering if it would be possible to add such a feature (bagfile splitting/cutting) as an enhancement of the software you provide (in addition to the tools) as it might be relevant to other users as well that do not want to go the ROS path.

MartyG-RealSense, 2020-02-14:
I've added the 'Enhancement' label to this post to tag it as a Feature Request that the RealSense developer team can track.

antoniogrimalt, 2024-06-07:
> @MartyG-RealSense thank you for your response. I will use the cut.py then or skip X frames. I was wondering if it would be possible to add such a feature (bagfile splitting/cutting) as an [...]

Hi @g2-bernotas, did you found a solution that produced .bag compatible with the RealSense Viewer?

closed 2026-10-25 by MartyG-RealSense