realsenseai/librealsense #5849: Modifying mapped_pixels value in kernel_transfer_pixels in cuda-align.cu

rb9175, 2020-02-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.11.15.0 | 
| Operating System & Version |  Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.15.0-36  | 
| Platform                                 | PC |
| SDK Version                            | 2.29.0 | 
| Language                            |  C++/cuda | 
| Segment			| Robot | 

### Issue Description

Trying to understand the alignment from depth stream to rgb stream, we weren't able to understand some pieces of code in cuda-align.cu. I hope you don't mind answering them.

In cuda-align.cu line 49 and 50:

mapped_pixels[mapped_index].x = static_cast<int>(other_pixel[0] + 0.5f);
mapped_pixels[mapped_index].y = static_cast<int>(other_pixel[1] + 0.5f);

What does the 0.5 at the end stand for? Is it rounding up to some pixel value? 

Also, at line 122:
new_val = new_val << 16 | new_val
What is happening here? Will new_val ever get changed?

Also, at line 123:
atomicMin(&arr[other_pixel_index / 2], new_val);
Why do we divide the index by 2?

Thanks!

RealSenseSupport, 2020-08-05:
About Line 49~50, It should be half round up.
https://www.mathsisfun.com/numbers/rounding-methods.html

About Line 122~123, the parameters' type of the function atomicMin is int (32-bit). but the depth value is 16-bit.
https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html#atomicmin

RealSenseSupport, 2020-09-04:
@rb9175 Did you get it through? Looking forward to your update. Thanks!

rb9175, 2020-09-11:
Yes, the issue is resolved. Thanks!

RealSenseSupport, 2020-09-16:
@rb9175 Glad to hear that. Close this issue accordingly. Thanks!

closed 2020-09-16 by RealSenseSupport