realsenseai/librealsense #5850: Strange X Axis Drift. Plot Attached

zainmehdi, 2020-02-15:
Camera Model | T265
-- | --
Firmware Version | 0.2.0.926
Operating System & Version | Linux (Ubuntu 18)
Kernel Version | 4.15.0-54-generic
Platform | NanoPI
SDK Version | 2.31
Language | C++
Segment | Robot

### Issue Description
I am having a strange drift in X axis even though I am moving in a straight line. I am attaching my plots. All units are in cm. I have turned off pose jumping and re localization. Kindly provide some insight. Is it normal ?
[image]

MartyG-RealSense, 2020-02-15:
@radfordi will be able to answer this question better than I can, but I will try to provide some initial input.  

There was a case where velocity was drifting off to the right even when the robot was stationary, and doing it during motion too.  They reported that it only occurred for them when pose jumping was set to false.

https://github.com/IntelRealSense/librealsense/issues/4876

zainmehdi, 2020-02-15:
Thanks @MartyG-RealSense I will try to turn pose jumping on and check the performance

msadowski, 2020-09-08:
Hi!

I was reporting this issue back in the day (https://github.com/IntelRealSense/librealsense/issues/4876). Unfortunately the issue still seems to be there, which is a pity. Here is how the odometry (position in the top tab and velocity in the bottom one) looks like for me after driving the robot for a bit:

[image]

This is with disabled pose jumping and taking wheel odometry from a ROS topic. I've double checked wheel odom and it provides 0 twist in all axes.

@MartyG-RealSense do you know if there is any work in the pipeline for T265 or is it going to be left as is?

MartyG-RealSense, 2020-09-08:
Hi @msadowski  I am not on T265 related cases now as somebody else on the RealSense team is responsible for those, so I cannot comment.  I recommend creating a new case to ask the question above so that the appropriate team member can respond.

msadowski, 2020-09-14:
@MartyG-RealSense many thanks for responding! Unfortunately it looks like T265 will not be attended to anymore: https://github.com/IntelRealSense/librealsense/issues/4518#issuecomment-691242254

RealSenseSupport, 2021-03-19:
Thank you for highlighting this issue. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265. There are some known drift that happens with the product. It is worth reviewing the specification update as outlined on the following page: https://dev.intelrealsense.com/docs/tracking-camera-t265

closed 2021-03-19 by RealSenseSupport