realsenseai/librealsense #5851: How to get D435 error log file from Intel Realsense Viewer?

Dontla, 2020-02-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       |

MartyG-RealSense, 2020-02-15:
First, open up the RealSense Viewer's "Settings" window by left-clicking on the gear-wheel icon in the top corner of the Viewer window and selecting 'Settings' from its drop-down menu.

[image]

When the Settings window pops up, left-click on the 'General' tab at the top of it.  Then left-click on the 'Output librealsense log to file' box to turn it blue.    The current path to the log file will be displayed.

[image]

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

Thanks MartyG!

Does this resolve your issue Dontla?

If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-03-10 by dorodnic