realsenseai/librealsense #5852: What could be causing 'Backend in rs2_create_device...'?

Dontla, 2020-02-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
Backend in rs2_create_device(info_list:00000268BD9F2E70, index:0):
hr returned: HResult 0x800703e3: "1
[image]
Then I can't find any of my cameras in the Intel Realsense Viewer, even if I have four connected...

MartyG-RealSense, 2020-02-15:
This error tends to occur when camera privacy settings are enabled in Windows.   Sometimes the privacy settings may get switched on automatically when Windows upgrades to a newer version.  

Please see the link below for information on how to make sure that windows is able to access the camera hardware and camera software applications.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

Did that solve your problem? Also noticed you're using an older version of RealSenseViewer. Always best to try on latest.

Thanks

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-04-05 by RealSenseCustomerSupport