realsenseai/librealsense #5854: What could be causing 'RT IC2 Config error'?

Dontla, 2020-02-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
[image]

log:
 15/02 23:05:37,709 INFO [11424] (rs.cpp:2160) Framebuffer size changed to 1366 x 705
 15/02 23:05:37,709 INFO [11424] (rs.cpp:2160) Window size changed to 1366 x 705
 15/02 23:05:37,709 INFO [11424] (rs.cpp:2160) Scale Factor is now 1
 15/02 23:05:38,453 INFO [11424] (tm-context.cpp:39) Failed to create TrackingManager
 15/02 23:05:38,640 INFO [10900] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "ЛŁ��"
 15/02 23:05:38,640 INFO [10900] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "ЛŁ��"
 15/02 23:05:38,640 INFO [10900] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "ЛŁ��"
 15/02 23:05:38,640 INFO [10900] (sensor.cpp:367) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]
 15/02 23:05:39,018 INFO [10900] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "ЛŁ��"
 15/02 23:05:39,018 INFO [10900] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "ЛŁ��"
 15/02 23:05:39,018 INFO [10900] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "ЛŁ��"
 15/02 23:05:39,018 INFO [10900] (win-helpers.cpp:86) _reader->GetNativeMediaType(sIndex, k, &pMediaType.p) returned: HResult 0xc00d36b3: "ЛŁ��"
 15/02 23:05:40,046 INFO [2396] (types.h:1071) RT IC2 Config error
 15/02 23:05:40,340 INFO [14608] (types.h:1071) RT IC2 Config error

MartyG-RealSense, 2020-02-15:
Are you using the official short USB cable that was supplied with the camera, please?

Dontla, 2020-02-16:
@MartyG-RealSense No, I use a two meter USB cable for each camera

MartyG-RealSense, 2020-02-16:
USB cable quality can vary greatly.  Ideally, cables used for RealSense that are 2 meters or longer should be high quality, otherwise performance may be degraded or the camera may not connect at all.  Are you able to test the camera with the official short cable please to test if the camera works okay with it?

Dontla, 2020-02-16:
@MartyG-RealSense I just checked the four cameras I plugged in and found that I said something wrong earlier. They are all connected using a 1m USB3.1 cable. Is there another reason that may cause RT IC2 config error?

MartyG-RealSense, 2020-02-16:
Do you have four cameras all plugged into the same computer?  

The only other thing to try for this error that I know of is to plug the camera into a mains electricity powered USB hub.  If you **are** using four cameras at the same time, I wonder if the computing device that they are plugged into does not have enough power to supply all of the cameras (something that a powered USB hub is meant to correct).

Do you get the error if using only one camera?

Dontla, 2020-02-18:
@MartyG-RealSense Yes, when I plug a D435 in a laptop, the Intel Realsense viewer will show RT IC2 Config Error. I don't know if this is the cause of insufficient power for that laptop. Now I want to use a USB hub that can supply power. Can you recommend this series of products to us? The number of cameras we use on a single computer may be between six and nine.
Best Regard!

MartyG-RealSense, 2020-02-18:
In Intel's multiple camera white paper, they use the 4-port AmazonBasics model in the link below:

https://www.amazon.com/dp/B00DQFGH80/ref=asc_df_B00DQFGH805315309/

Amazon do a 10-port AmazonBasics model if you prefer to keep everything centralized on one hub.

https://www.amazon.com/AmazonBasics-USB-10-Port-Power-Adapter/dp/B076YN6CSG

With hubs, there is a '5 deep' rule in the USB standard that allows connection of up to 5 hubs to a computer to provide the number of USB ports that you need.

RealSenseCustomerSupport, 2020-03-09:
----------------------------------------------

Hi Dontla,
Is this still an issue for you after you use powered hub or did any change?

RealSenseCustomerSupport, 2020-04-17:
----------------------------------------------

**If we don’t hear from you in 7 days, this issue will be closed**

closed 2020-04-20 by RealSenseCustomerSupport

rajeshmanpat, 2021-04-13:
Facing same issue with standard cable that came with the camera from Intel. D455, Windows/Ubuntu 18, all the same issue.

MartyG-RealSense, 2021-04-13:
Hi @rajeshmanpat   Could you create a new case for your problem please using the green **New Issue** button on the front page of the forum at the link below?  Thanks!

https://github.com/IntelRealSense/librealsense/issues/