realsenseai/librealsense #5856: nodejs wrapper broken.

gavan1, 2020-02-16:
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   { MacOS  Catalina}| 
| SDK Version                            |  {  2.0 }                          | 
| Language                            |  {nodejs }                          | 
| Segment			|  {VR/AR }                          | 

node v8.17.0. ...when using a more recent  version 12, node wrapper would not build.

running node nodejs-capture.js
```
` Error: Unrecoverable! error native function rs2_pipeline_wait_for_frames: Frame didn't arrived within 5000`

 it detect when device is not connected which is a good sign.
`Error: Unrecoverable! error native function rs2_pipeline_start: No device connected`

cmake Realsense viewer tool runs correctly.

my guess, nodejs wrapper just needs to be remade with current node version.

gavan1, 2020-02-16:
@ev-mp does this this sound like I should be downgrading my firmware D435i?

dorodnic, 2020-02-17:
Hi @gavan1 
Building node.js wrapper from source should be an option, and we check it is not broken with every commit ([example build](https://travis-ci.org/IntelRealSense/librealsense/jobs/650812048?utm_medium=notification&utm_source=github_status))

gavan1, 2020-02-20:
@dorodnic on your CI build end I'm sure its building... but on the end to end testing... quite broken. Has there been any documented examples by the community of a working nodeJS wrapper? Intel needs to increase your team lol.

dorodnic, 2020-02-26:
Hi @gavan1 
Sorry for that, I'll try to run it on a Mac end-to-end. 
I must say that community interest in the node.js wrapper has been disappointingly limited, while it being one of the more complex wrappers to develop and maintain :(

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

"""
D435i is supported on Linux, Windows 10 and Android. Mac OS support is scheduled next, and Windows 7 will be provided in future releases. requires FW 5.11.6.250+.
"""

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

Sorry. I notice something about D435i on macOS in release note.
"""
D435i is supported on Linux, Windows 10 and Android. Mac OS support is scheduled next, and Windows 7 will be provided in future releases. requires FW 5.11.6.250+.
"""

RealSenseSupport, 2020-12-28:
@gavan1 PR https://github.com/IntelRealSense/librealsense/pull/7573 might resolve your issue and is included in latest SDK v2.41. Could you please have a try? Thanks!

RealSenseSupport, 2021-01-15:
@gavan1  Did you get chance to try SDK v2.41? Looking forward to your update. Thanks!

RealSenseSupport, 2021-05-25:
@gavan1 Could you please update? Please note that the issue will be closed  if we don't hear from you for another 7 days. Thanks!

closed 2021-08-17 by maloel