realsenseai/librealsense #5857: Is it  posible add ICP odometry fusion from a 2D lidar on T265?

FPSychotic, 2020-02-16:
I think it could be nice to substitute/complement the wheel odometry fusion with the ICP odometry from a 2D lidar, maybe it could be useful too for relocalization, calibration, loop closure...

MartyG-RealSense, 2020-02-16:
ICP odometry is supported by RTABMAP, software which has been used a number of times with a pairing of T265 and a 400 Series depth camera. The link below discusses though why ICP is disabled by default on RTABMAP due to it being a negative influence in some use-cases.

https://github.com/introlab/rtabmap/wiki/ICP

If you wanted to pair the T265 with the forthcoming L515 lidar depth camera, I would speculate that it is a possibility.  At the time of the T265's launch, @dorodnic said that "this device can be used with RealSense (and most likely any 3rd-party) depth cameras".

FPSychotic, 2020-02-16:
Thank you by the advice, I didn't have acknowledged of that situation. My setup os d435+t265+2D 360 Lidar and I don't how I should integrate it with the cameras. I though that maybe it could substitute the use of wheel encoders and get at least the same improvement than could get with cheap en encoders or from a VESC and its sensored motor. I can use Rtabmap, or any other software.

MartyG-RealSense, 2020-02-16:
Do you need further advice on this subject, please?

closed 2020-02-16 by FPSychotic