realsenseai/librealsense #5859: Wrong /camera/aligned_depth_to_color/image_raw

drfg0119, 2020-02-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300 / D400 } | 
| Firmware Version                   |   | 
| Operating System & Version |   Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    | 4.9                                      | 
| Platform                                 | NVIDIA Jetson TX2 (Jetpack4.2) |
| SDK Version                            |  {  2.31.0 & 2.22.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  Robot                          | 

### Issue Description
Hi, I got a problem about performance of SR305, when I use TX2 to connect it on the API(realsense-viewr_2.31.0), I didn't see the results as same as when I connect on PC. The working distance isn't that good on TX2, when I turn on the Coded-Light Depth Senser, the working distance is almost only 10 cm, and the depth of pixel is inaccurate.
Btw, the SDK version on TX2 I used is the same on PC.

drfg0119, 2020-02-18:
It's the aligned depth to color image on PC.
[image]
And this is the incomplete frames on TX2.
[image]
But there is no problem when I run the API on TX2.
[image]

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

Hi @drfg0119,

Questions related to ROS should be asked on a forum that intended for ROS  questions:
https://github.com/IntelRealSense/realsense-ros/issues
Please close the issue here and open it on above-mentioned forum.

Thank you!

drfg0119, 2020-02-27:
@RealSenseCustomerSupport Hi, OK!! Thank you !

closed 2020-02-27 by drfg0119