realsenseai/librealsense #5861: Hardware Sync

nawara72, 2020-02-17:
Hi Just want to know if the following information are true or not?

Only hardware depth sync is available on D435, not color syn. Is it true that we can only hardware synchronize the depth images but not the color images from multiplke D435s??

Auto exposure for the depth camera D435 has to be off if hardware sync is needed?

Thanks

MartyG-RealSense, 2020-02-17:
A past answer provided by. @dorodnic is: "D435 allows Depth to Color sync, and Depth to another Depth sync, but not Color to another Color, so at least one (or both) of the color sensors will not be synced".

This comment is sourced from an excellent discussion on hardware sync that is worth reading.

https://github.com/IntelRealSense/librealsense/issues/2637

Regarding auto-exposure: the common advice seems to be to disable **Auto-Exposure Priority** during hardware sync.  This setting is not the same thing as Auto-Exposure.

https://github.com/IntelRealSense/librealsense/issues/2637#issuecomment-434203140

Typically when Auto-Exposure Priority is disabled, Auto-Exposure should be left enabled.  The reason for disabling Auto-Exposure Priority is usually to enforce a constant FPS.

nawara72, 2020-02-17:
Thanks MartyG for the info. I guess my next question is what frame rate do I set the D435 to if I want to asynchronously externally trigger it? any other info you can provide on asynchronous hardware triggering would be greatly appreciated. Thanks

agrunnet, 2020-02-17:
Currently there is no asynchronous triggering feature. You want to trigger of an external signal of any frequency, correct?  This feature will come but right now you stand different cameras to exactly the same frequency determined by the master camera (being a Realsense camera).

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

Is there any more question?

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

nawara72, 2020-02-25:
That's all for now :-) Thanks for all the help

closed 2020-02-25 by nawara72