realsenseai/librealsense #5863: libgtk-3-dev has unmet dependencies or sources.list is corrupt.

Naba7, 2020-02-17:
https://github.com/IntelRealSense/librealsense/issues/3772#issue-433718328 I am facing the similar issue and can't resolve it.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi @Naba7,

Please describe the exact scenario that leads to the issue. Do you use an officially supported Ubuntu distro and kernel with X11 server?

Thank you!

RealSenseCustomerSupport, 2020-03-03:
----------------------------------------------

@Naba7,

Do you still need help with this question?

closed 2020-03-10 by dorodnic

dnabanita7, 2020-11-27:
Hi @RealSense-Customer-Engineering ,

I want to install dlib package but for that I have to install the dev package of libgtk-3 . It is showing error as: 

```
The following packages have unmet dependencies:
 libgtk-3-dev : Depends: libatk-bridge2.0-dev but it is not going to be installed
                Depends: libatk1.0-dev (>= 2.15.1) but it is not going to be installed
                Depends: libcairo2-dev (>= 1.14.0) but it is not going to be installed
                Depends: libfontconfig1-dev but it is not going to be installed
                Depends: libfribidi-dev (>= 0.19.7) but it is not going to be installed
                Depends: libgdk-pixbuf2.0-dev (>= 2.30.0) but it is not going to be installed
                Depends: libglib2.0-dev (>= 2.57.2) but it is not going to be installed
                Depends: libpango1.0-dev (>= 1.41) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

I use official Ubuntu distro.