realsenseai/librealsense #5864: Could not initialize offscreen context

mkeyno, 2020-02-17:
I bought an intel realsense depth camera D435i and trying to install realsense .
I exactly followed the tutorial
When I am running realsense-viewer , I am getting the following error :
Could not initialize offscreen context!
----------------------------------------------------------------------------------------------------
hwinfo
----------------------------------------------------------------------------------------------------
graphics card:                                                  
                       Acer Incorporated [ALI] GeForce 710M
                       Intel 3rd Gen Core processor Graphics Controller

Primary display adapter: #14
-Computer-
Processor		: Intel(R) Core(TM) i3-3120M CPU @ 2.50GHz
Memory		: 3855MB (2398MB used)
Machine Type		: Notebook
Operating System		: Linux Lite 4.8
-Version-
Kernel		: Linux 4.15.0-74-generic (x86_64)
Version		: #84-Ubuntu SMP Thu Dec 19 08:06:28 UTC 2019
C Library		: GNU C Library / (Ubuntu GLIBC 2.27-3ubuntu1) 2.27
Distribution		: Linux Lite 4.8

-Display-
Resolution		: 1366x768 pixels
OpenGL Renderer		: (Unknown)
X11 Vendor		: The X.Org Foundation
-Audio Devices-
Audio Adapter		: HDA-Intel - HDA Intel PCH
-Input Devices-
 HDA Intel PCH Mic
 HDA Intel PCH Headphone
 HDA Intel PCH HDMI/DP,pcm:3
 HD WebCam: HD WebCam
 Intel(R) RealSense(TM) Depth Ca
-Display-
Resolution		: 1366x768 pixels
Vendor		: The X.Org Foundation
Version		: 1.19.6
Current Display Name		: :0.0
-Monitors-
Monitor 0		: 1366x768 pixels
-OpenGL-
Vendor		: (Unknown)
Renderer		: (Unknown)
Version		: (Unknown)
Direct Rendering		: No
-Extensions-
Composite
DAMAGE
DOUBLE-BUFFER
DPMS
DRI2
GLX
Generic Event Extension
MIT-SCREEN-SAVER
MIT-SHM
Present
RANDR
RECORD
RENDER
SECURITY
SGI-GLX
SHAPE
SYNC
X-Resource
XC-MISC
XFIXES
XFree86-DGA
XFree86-VidModeExtension
XINERAMA
XInputExtension
XKEYBOARD
XTEST
XVideo
default screen number:    0

dorodnic, 2020-02-17:
Hi @mkeyno 
This error is not related to the RealSense camera - you should be able to run most examples and use the API. Viewer requires OpenGL 3 support, hence the error. It seems you don't have graphic drivers correctly installed, in particular:
```
-OpenGL-
Vendor : (Unknown)
Renderer : (Unknown)
Version : (Unknown)
```

mkeyno, 2020-02-21:
@dorodnic 
thanks for your comment, I reinstall driver with other method and software but no success 
I think this repo is not mature or hard to implement for Linux base  OSes, because I also couldn't successful using my camera over raspberry pi 4
I kinda regret to buy this product

dorodnic, 2020-02-24:
Sorry you feel this way.
Not sure what having OpenGL drivers installed has to do with the product or repo maturity

closed 2020-02-24 by dorodnic

Kosta404, 2020-07-03:
@dorodnic
_realsense-viewer_ runs smoothly, when raspberry pi 4 is connected with HDMI to a monitor. But through VNCserver gives output **"could not initialize offscreen context"** . Help sovling th problem please.