realsenseai/librealsense #5865: calculating the distance between object to the depth camera.

srividya91, 2020-02-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  {2.3.10} | 
| Language                            | {python} | 
| Segment			|  {Robot } | 

Hello Everyone. i am working on intel real sense D435i and i want to know how to calculate the distance between the detected object to the depth camera.. can any one suggest me how to do that?

jb455, 2020-02-17:
This example should get you there: https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

ShettyHarapanahalli, 2020-02-17:
this could be of use for you - 
https://github.com/IntelRealSense/librealsense/issues/5779

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

Hi @srividya91
Do you need further help on this?
If we don't hear from you in 7 days, this issue will be closed.

closed 2020-03-12 by RealSenseCustomerSupport