realsenseai/librealsense #5868: usb webcam broken after update to librealsense2-dkms 1.3.13-0ubuntu1

franzabist, 2020-02-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  5.3.0-40-generic                                     | 
| Platform                                 | Dell G3 3590 i7-9750H, 16GB, GTX 1650  |
| SDK Version                            |  v2.32.1 /  librealsense2-dkms 1.3.13-0ubuntu1  }                          | 
| Language                            |  C++                           | 
| Segment			|  Robot                           | 

### Issue Description - Bug: Interference with internal / external USB web cams

After upgrading realsense sdk from v2.31.0 to v2.32.1, the internal laptop web cam as well as
externally attached USB web cams disappear from  /dev/video0, /dev/video1, ...
and cannot be accessed anylonger by software such as cheese or ros usb_cam.

After manually downgrading librealsense2-dkms from version 1.3.13-0ubuntu1 to 1.3.12-0ubuntu1,
both the internal and external webcams immediately re-appear in  /dev/video0, /dev/video1, ... and work in applications.

The bug happens only on the Dell G3 3590 which has both USB 3 and USB 2 ports. 
The bug cannot be reproduced on a MacBook Air 2017 which has only USB 3 ports.

ev-mp, 2020-02-18:
@franzabist , I suspect that you have encountered an issue with trying to use the patched driver, such as Secure Boot. #5598, #5231, #5631.
Can you provide the kernel log for confirmation?

Regarding DKMS versions: v1.3.13 is the first to provide kernel 5.3 support. Installing the previous versions would not insert kernel patches.

franzabist, 2020-02-21:
Please find the kernel log below.
Realsense T265 works fine after installation of librealsense2-dkms version 1.3.13-0ubuntu1 both in  realsense-viewer as well as in ROS.
However, neither the internal laptop webcam nor other externally attached USB webcams are available any longer. 

The system is installed on an external USB-drive. When attaching it to an MacBook Air 2017, this problem does not occur and USB web-cams can still be used.
The problem occurs also on another Ubuntu system which is on kernel 5.0.0-37-generic.
 
> sudo apt-get install librealsense2-dkms

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfwup1
Use 'sudo apt autoremove' to remove it.
The following packages will be upgraded:
  librealsense2-dkms
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 10.2 MB of archives.
After this operation, 22.5 MB of additional disk space will be used.
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 librealsense2-dkms all 1.3.13-0ubuntu1 [10.2 MB]
Fetched 10.2 MB in 4s (2,618 kB/s)             
(Reading database ... 321575 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.13-0ubuntu1_all.deb ...

------------------------------
Deleting module version: 1.3.12
completely from the DKMS tree.
------------------------------
Done.
Unpacking librealsense2-dkms (1.3.13-0ubuntu1) over (1.3.12-0ubuntu1) ...
Reloading original modules into kernel... complete
Current status:
bcmwl, 6.30.223.271+bdcom, 4.15.0-74-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 4.15.0-88-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 5.3.0-28-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 5.3.0-40-generic, x86_64: installed
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d 	
Setting up librealsense2-dkms (1.3.13-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.13 DKMS files...
Building for 5.3.0-40-generic
Building initial module for 5.3.0-40-generic
Done.

uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

hid-sensor-gyro-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

depmod...

DKMS: install completed.

Loading the modified modules into kernel...modprobe: ERROR: could not insert 'hid_sensor_gyro_3d': Operation not permitted
Current status:
bcmwl, 6.30.223.271+bdcom, 4.15.0-74-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 4.15.0-88-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 5.3.0-28-generic, x86_64: installed
bcmwl, 6.30.223.271+bdcom, 5.3.0-40-generic, x86_64: installed
librealsense2-dkms, 1.3.13, 5.3.0-40-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.13
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms

closed 2020-02-21 by franzabist