realsenseai/librealsense #5869: Function call in comment line

militaryCoder, 2020-02-18:
Message sent on [this line](https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/CMake/connectivity_check.cmake#L2) would not be ever shown.

MartyG-RealSense, 2020-02-18:
Thanks for the report!  I'll just clarify for others reading this that there is a link to a line of code, as the link was hard to see at first.

https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/CMake/connectivity_check.cmake#L2

[image]

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi @militaryCoder,

Thank you for pointing this out. All necessary corrections will be made as soon as possible.

RealSenseSupport, 2020-10-09:
@militaryCoder The corrections now merged as https://github.com/IntelRealSense/librealsense/blob/master/CMake/connectivity_check.cmake#L3 . Thanks for pointing this out!

MartyG-RealSense, 2020-10-09:
@RealSenseSupport   Thanks for the update about the corrections   :)

@militaryCoder  I will close this case now that the corrections have been implemented.   Thanks again for your report!

closed 2020-10-09 by MartyG-RealSense