realsenseai/librealsense #5870: manual_reset_event    use wrong parameter

KennyYaming, 2020-02-18:
src\win\win-helpers.cpp    lines  634

        manual_reset_event::manual_reset_event()
            :event_base(CreateEvent(nullptr, **FALSE**, FALSE, nullptr))
        {}

 I think the second parameter should be  TRUE.

dorodnic, 2020-02-18:
Hi @KennyYaming 
I think you are right, thank you

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

Hi @KennyYaming,

Thank you for pointing this out. All necessary corrections will be made as soon as possible.

RealSenseSupport, 2020-10-09:
@KennyYaming The fix now was merged to Release. https://github.com/IntelRealSense/librealsense/blob/master/src/win/win-helpers.cpp#L792 Thanks again for your pointing out this!

closed 2020-10-12 by RealSenseSupport