realsenseai/librealsense #5871: 3d points from 2 D435 camera are different

svabhishek29, 2020-02-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.02.100) | 
| Operating System & Version |   {Win (10)|                                          | 
| Platform                                 | PC|
| SDK Version                            |  { 2.32.1.1299 }                          | 
| Language                            |  {python}                          | 

we have 2 intel D435 Cameras. we initially used one camera to build the entire AR application. when it was working well we connected the second camera and started noticing that the 3D points given by the 2nd camera are different compered to the 1st camera. how can this be fixed

i don't know if this is a important, but the first camera was bought in 2018 nov and the other one was bought recently a couple of months back

Thanks

dorodnic, 2020-02-18:
What do you mean by different? (average distance? plane-fit RMS?)

svabhishek29, 2020-02-18:
there is an offset that we are seeing in the new camera. we are using vtk to create an AR environment.
in that when we place a sphere in vtk, with the first camera it place on the point on interest in the second camera the sphere is to the left and top of the point of intrest

dorodnic, 2020-02-18:
What order of magnitude of the shift? 
You can also use Depth Quality Tool to gather some quantitive statistics from both cameras.
Camera getting out of calibration (due to physical damage during shipment) can result in higher plane-fit RMS or absolute offset. We provide tools (dynamic calibration and on-chip calibration) that can bring camera back to calibration.

svabhishek29, 2020-02-24:
when i select the middle pixel for example on a 640 x 480 image, the middle pixel is 320, 240. The 3D point should be at 0, 0, z value but i am getting 0.010891272686421871, -0.01863032765686512, z value. how can this be fixed?

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

Based on your description of the issue, it sounds like the 2 cameras have a slight x,y offset. By itself, I don't think this is surprising since the principal (center) points (PP) for each camera are generally different.

But, let me try to confirm this with some follow-up questions:
Is the offset you see constant over time or do you find that it changes?
For a given calibration, this offset should not change or not by much, unless the camera undergoes some physical change (shock, vibration, large temperature variation).
If you do re-calibrate the camera, it is possible for it to change the PP.
Is your setup repeatable - is the 1st camera always at (0,0)? If so, then I would assume the 2nd camera always has the same offset.

Assuming the offset is stable, then you should be able to apply this slight shift of the reference coordinates for each additional camera. Similarly, if the Z values for a particular object are slightly different (and also stable), you could apply a small scaling factor to the others, keeping one as the reference.

Let us know if this addresses the question or if you're seeing something else.

RealSenseCustomerSupport, 2020-05-15:
----------------------------------------------

Hi @ svabhishek29 Do you need further help on this?
If we don't hear from you in 7 days, we will close this issue.

closed 2020-05-29 by RealSenseCustomerSupport