realsenseai/librealsense #5872: Copy-paste in documentation

militaryCoder, 2020-02-18:
Copy-paste typo, I think.
https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/CMakeLists.txt#L14
https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/CMakeLists.txt#L20

MartyG-RealSense, 2020-02-18:
It looks as though the note is referring to two separate internet checks for the tracking module support and for the depth camera.  So it is valid to have the message for each of the checks, though the comment could perhaps be edited to provide a bit of extra clarity.  I've added the 'documentation' label to your message in case somebody on the developer team wants to edit those lines in the code.

RealSenseCustomerSupport, 2020-04-05:
----------------------------------------------

https://github.com/IntelRealSense/librealsense/pull/6200

Thanks militaryCoder!

closed 2020-04-24 by RealSenseCustomerSupport