realsenseai/librealsense #5873: Accuracy of using t265 to track april-tags

talbenach, 2020-02-18:
We are considering using a stationary t265 camera in an application that tracks tools marked with April-tags from distances that range between 30cm and 80 cm. The April-tags are approximately 2cm squares.
What is the expected position accuracy of this setup? What factors affect this accuracy?
Thank you very much

dorodnic, 2020-02-18:
Hi @talbenach 
There is no official accuracy metrics we can share. 
Please note that april-tag tracking is currently provided as an example and done on the CPU, and not on device. Hence, I'm not sure if there is any benefit to using t265 if the camera is stationary.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi @talbenach

Is there anything else needed on this thread?

***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-05 by RealSenseCustomerSupport