realsenseai/librealsense #5874: T265 ASIC role

talbenach, 2020-02-18:
The T265 datasheet states that it equipped with an ASIC for on board calculations. I could not find details on what these calculations are and what is the output of the camera to the computer.

MartyG-RealSense, 2020-02-18:
In the context of the T265 data sheet, the ASIC in the T265 is the Movidius Myriad 2 VPU.  Below is a quote and image from the official T265 promotional information page.

***********

The Intel Movidius Myriad 2 VPU is a system-on-chip component that is purpose-built for image processing and computer vision at very high performance per watt, including for space-constrained implementations. Its key architectural features are the following:

• Vector processor cores are optimized for machine vision workloads.
• Hardware accelerators increase throughput for imaging and computer vision.
• General-purpose RISC CPU cores coordinate and direct workloads and interaction with external systems.

[image]

 To quote Intel's guide to getting IMU data:

***************

T265 redirects IMU readings into an Intel Movidius Myriad 2 Vision Processing Unit (VPU). The inertial sensor data is also complemented by video from two fisheye monochrome sensors that are fed into the VPU as well.

The VPU employed with T265 is capable of fusing inertial and video sensors to produce a reliable and persistent six degrees of freedom (6DOF) pose, offloading the CPU-intensive calculations from the host PC.

************

This processing of data in the T265's onboard Movidius VPU is different from the IMU implementation on the D435i, which delivers the raw IMU data directly to the host PC,

talbenach, 2020-02-18:
It appears that the VPU is used to calculate the camera’s 6DoF. Does it play any role in tracking? Can we access the VPU to aid in tracking and reduce load from the host PC?

[image]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F41145062%2F74752886-3dca5200-5267-11ea-9d6c-734a179e470f.png&data=01%7C01%7Cnoaml%40brainsgate.com%7Cd141446564454a25770608d7b48bafff%7Cba1b9a8211954a63b7107dab2a24ddaf%7C0&sdata=Ju%2Bx6SGDhGyLc9JQF4gEGlgYmpRQ3%2BImV0cdij6VObA%3D&reserved=0>

MartyG-RealSense, 2020-02-18:
As far as I know, all the data on the T265 is processed through the Movidius VPU by default.  The link below has an official Intel case-study for using the T265 that describes the VPU's role in tracking in the second to last section near the bottom of the page.  The section is headed "Host connectivity and integration into devices".

https://www.intelrealsense.com/visual-inertial-tracking-case-study/

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

To add to MartyG-RealSense, T265 is a closed system SLAM solution.  The Movidius VPU is used to take input from the IMU that is onboard and the fisheye cameras and utilize an Intel proprietary SLAM algorithm to calculate it's position in space.  The final output to host system is pose data.  One can also get raw fisheye images and raw IMU (accel and gyro) data to the host system for host side processing if developer so needs that information.  We do not allow for developers to use the Movidius VPU to run their own FW/Algorithm.  T265 runs it's own FW/Algorithm.

Let us know if you need anything else on this topic.

talbenach, 2020-02-23:
Thank you very much for this clarification. I assume the answer to the following is no, but I want to make sure I am not missing anything. Is there access to feature recognition data generated by the Movidius? E.g. corners recognition?

Thank you,

MartyG-RealSense, 2020-02-23:
Intel's guide to Visual SLAM and the T265 has the following quote:

***********

This device is intended for customers looking for a hassle-free stand-alone SLAM system.  **For those who want to develop or deploy their own tracking algorithms, they can obtain the raw data (cameras and IMU) from the T265**, or build on top of our Intel RealSense Depth Camera D400 family which provide similar sensor capabilities (albeit with smaller FOV) and also output depth maps. 

However, note that any SLAM algorithm would need to be run on the host platform and not embedded and this will increase the weight, power budget and cost of the target platform.

***********

The Visual SLAM guide is available at the link below.

https://dev.intelrealsense.com/docs/intel-realsensetm-visual-slam-and-the-t265-tracking-camera

It discusses feature detection algorithms such as corner detection in a general way not specific to the T265, quoting algorithms such as Harris Corner and Shi-Tomasi Corner.  The image below from the guide shows corner detection using the Harris Corner algorithm,

[image]

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

No access to feature data from the Movidius integrated in T265.

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Is there anything else needed on this thread?

**If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by dorodnic