realsenseai/librealsense #5875: How to get depth preset setting of recorded video using pyrealsense2

anguyen216, 2020-02-18:
Hi,

I recently used D435i to recorded some video data.  I used different depth preset setting for different set of videos.  But now I lost track of which preset I used for which set.  I want to know if there's any function in pyrealsense2 to extract depth preset setting of videos (or just individual depth frame).  If there is, please show me an example of how to use the function.  

Thanks!

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Operating System & Version |   {Win (8.1/10)}  | 
| Platform                                 | PC  |
| Language                            |  {python }                          | 

### Issue Description
<How to get depth preset setting of recorded video>

dorodnic, 2020-02-18:
If you drop the bag file into the viewer. Everything that was recorded will be in the device panel. 
I'm not 100% sure if all of the advanced controls are recorded, but if you selected particular preset it would appear as Visual Preset control

anguyen216, 2020-02-18:
> If you drop the bag file into the viewer. Everything that was recorded will be in the device panel.
> I'm not 100% sure if all of the advanced controls are recorded, but if you selected particular preset it would appear as Visual Preset control

Thank you! I didn't know I could do this.  But I was also hoping that there is a function that can do this so I could extract preset of bag file quickly.  I have over 100 videos

dorodnic, 2020-02-18:
This should be similar to regular device in python - get device from pipeline start, go over sensors, go over supported options, if rs.options.visual_preset is supported, read its value.

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@anguyen216  Did you get it through with the reply from dorodnic? Any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@anguyen216 Will close it if no other questions. Thanks!

closed 2020-03-23 by RealSenseCustomerSupport