realsenseai/librealsense #5876: map reuse with T265

3p3r, 2020-02-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Language                            |  {C++ }                          | 

### Issue Description

Hello!

We are trying to use the T265 camera in a medical trial where we have to use a pre-existing map (captured with T265) and cannot do map building every time. We just need the localization data after map building is done. I looked at the SDK APIs and found `rs2_import_localization_map` and `rs2_export_localization_map` for T265. Is this the way to achieve map reuse with T265?

If yes, is this possible with RealSenseViewer or we have to roll our own code with the SDK?

MartyG-RealSense, 2020-02-18:
I hope the link below will be useful.   It discusses exporting a map from the RealSense Viewer and re-importing it.

https://github.com/IntelRealSense/librealsense/issues/4526

3p3r, 2020-02-18:
@MartyG-RealSense I have seen that link before opening this issue, not sure how that relates to what I am trying to accomplish here. That link talks about correlating between two maps, whereas, I want to use a single map across multiple localization runs. I do not want to recapture the map again.

MartyG-RealSense, 2020-02-18:
I recommend looking at the **ar-advanced** sample program added recently to see if it meets your needs in regard to localisation map import / export.

https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-advanced

If that does not help, a T265 expert on the RealSense team such as @radfordi or @ev-mp  may be able to advise you.

3p3r, 2020-02-18:
I did look at that example. It's _almost_ what I want to accomplish, however T265 modifies the map I load into it. I want the camera to "freeze" the loaded map and just do localization inside it without modifying the loaded map.

radfordi, 2020-02-18:
Hi @3p3r, We've though about doing a mode where the map is fixed, but we don't have something like that at this time.

3p3r, 2020-02-18:
@radfordi thanks for the clarification. given the application in a medical trial, having the map regenerated every time adds randomness which is not ideal.

If map fixing is not possible, is there a way to quantitatively measure this randomness so we can report that on our paper? For example, if we load a pre-captured map and T265 modifies it, we export it again, is there a way to capture the diff between the two maps?

How would you suggest approaching this with T265?

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @3p3r, as outlined by radfordi, this is not something that is available at this time.
Let us know if there is anything else on this topic.

***If we don’t hear from you in 7 days, this issue will be closed.**

3p3r, 2020-03-10:
We ended up doing our trials all in one-go to reduce the randomness of building a map every time we do a trial. We wanted to load a map into the device to fix that variable but since the feature is absent, we just captured SLAM output continuously.

sinaaghli, 2020-04-15:
I also had a related question on how maps are handled in T265. 

Isn't map part of the pipeline something that users would want to maintain? for example in a larger scale map one might want to load only a section of the map that the camera is in and then as the map updates get detected those updates would be applied to the original map. 
I can easily see that in a multi-agent system, A unified map (stored on cloud) would be very much desired and each agent could load and apply map changes as they desire. 
With the current setup of the pipeline users have no way to manage the map sadly unless I'm missing a part of the whole story.

sinaaghli, 2020-04-15:
BTW In my opinion, you guys have created the most advanced sensor in the world (algorithmically), As a SLAM user I Admire all these efforts.

neilyoung, 2020-04-15:
I second that 100%. There is already a partial approach available, so it is possible to save and load and reuse and improve, but it is not there where it should be. Especially not if it comes to an on the fly management of maps, e.g. while passing building or floor boundaries. I don’t know if you know my results , but those completely underline what you said with respect to the sensor’s and algorithm’s quality. But just a few seem to know. IMHO the T256 is one of the most underestimated sensors nowadays.

https:// neilyoung.serveblog.net

neilyoung, 2020-04-15:
Besides that (I don't know exactly, since I never played with this option): Wouldn't be the `ENABLE_MAPPING` option exactly do, what is required? I mean, just from the name...

closed 2021-03-17 by RealSenseSupport