realsenseai/librealsense #5877: IMU Calibration will not load to camera

g-robot-tech, 2020-02-18:
I am having an issue with the rs-imu-calibration.py script. I can run through all of the six orientations and gather data for them. Then I can save the data. However, after all of that I never get the prompt of 'upload to camera?'

I only get a line that says 'array too big.'

Does anyone know how to fix this?
Thanks.

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

Hi @g-robot-tech

I've included a link below to the documentation around the imu calibration example for the D435i camera.  Let us know if you've reviewed this document as part of your process of utilizing the rs-imu-calibration script.  Also have you updated the camera FW to the latest FW?  You can use the realsense viewer that is part of the latest librealsense 2.32.1 to update the FW to the latest

https://dev.intelrealsense.com/docs/depth-camera-d435i-imu-calibration

If you can provide a bit more details of your configuration and setup, that would help.

g-robot-tech, 2020-02-22:
Hi,

Thank you for your response!

I have been using the documentation attached. I have updated the camera to the latest firmware of 2.32.1.

I have the D435i. Right now I have it in its box with a small hole cut for the connecting usb cord to exit. This is so I can steadily hold it at 90 degree angles.

 I have tried to run the rs-imu-calibration.py on both python2 and python3. I go through the whole process of holding the camera in the six different positions, I save the data, then I get the message below:
______________________________________________________________________________
Status.collect_data[....................]

Direction data collected.
Would you like to save the raw data? Enter footer for saving files (accel_<footer>.txt and gyro_<footer>.txt)
Enter nothing to not save raw data to disk. >test3

Writing files:
accel_test3.txt
gyro_test3.txt

Done. too many indices for array
______________________________________________________________________________

I believe I should be prompted with some sort of "would you like to upload this to your camera?" message, but I never receive that message.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi

Do you by chance have another D435i device to test with?  This particular issue haven't been seen before on our end.

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi

Do you by chance have another D435i device to test with?  This particular issue haven't been seen before on our end.

**If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport

Gutlapalli-Nikhil, 2025-02-04:
Hi.  I am encountering the same issue actually.

waiting for realsense device...
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  313522071178
  Product Line:  D400
  Firmware version:  5.13.0.55
Start interactive mode:
FOUND ACCEL with fps=100
FOUND GYRO with fps=200
-------------------------
*** Press ESC to Quit ***
-------------------------

Align to direction:  [ 0. -1.  0.]   Mounting screw pointing down, device facing out
 Status.collect_data[....................]                                                

Direction data collected.
Align to direction:  [1. 0. 0.]   Mounting screw pointing left, device facing out
 Status.collect_data[....................]                                                

Direction data collected.
Align to direction:  [0. 1. 0.]   Mounting screw pointing up, device facing out
 Status.collect_data[....................]                                                

Direction data collected.
Align to direction:  [-1.  0.  0.]   Mounting screw pointing right, device facing out
 Status.collect_data[....................]                                                

Direction data collected.
Align to direction:  [ 0.  0. -1.]   Viewing direction facing down
 Status.collect_data[....................]                                                

Direction data collected.
Align to direction:  [0. 0. 1.]   Viewing direction facing up
 Status.collect_data[....................]                                                

Direction data collected.
Would you like to save the raw data? Enter footer for saving files (accel_<footer>.txt and gyro_<footer>.txt)
Enter nothing to not save raw data to disk. >

Not writing to files.

Error: too many indices for array: array is 1-dimensional, but 2 were indexed
FATAL: exception not rethrown
Aborted (core dumped)

MartyG-RealSense, 2025-02-05:
Hi @Gutlapalli-Nikhil  https://github.com/IntelRealSense/librealsense/issues/12546#issuecomment-1872539479 has an example of saving the raw data to disk in the IMU calibration tool by inputting **20dec** when requested to enter a footer so that files called 'accel_20dec.txt' and 'gyro_20dec.txt' are saved.

Or did the error occur because you didn't enter anything because you did not want to save the data, please?

This error seems to be very rare as I could only find one other case other than your own where it has occurred in the past.

Gutlapalli-Nikhil, 2025-02-05:
Hi @MartyG-RealSense , thanks for the response. I even tried to save the file, still getting the same error. Not sure if the versions I am using are mismatched. 

I didn't go deep to debug the script yet as it worked for many people. Maybe I will try to give a shot by calibrating in windows, instead of linux and see if I face similar issue. Any suggestions would be greatly appreciated.

[image]

MartyG-RealSense, 2025-02-05:
In my opinion the IMU calibration tool is not as important as it used to be.  This is because there is a feature called Motion Correction that can 'fix' raw IMU data to improve its accuracy if it does not have good accuracy, but it used to be locked by default and could only be enabled once a calibration had been successfully written to the camera hardware.

These days though, the Motion Correction function is unlocked by default on RealSense cameras equipped with an IMU.  In the RealSense Viewer tool you can find it by expanding open the Motion Module category of the Viewer's options side-panel and then expanding open the Controls sub-category in that section.   It is enabled by default in the Viewer when an IMU is present.

[image]

Gutlapalli-Nikhil, 2025-02-06:
Cool, thank you. With the viewer, I am able to collect the data and process it, also its matching to the actions I am doing. So I am skipping the calibration on linux for now.