realsenseai/librealsense #5878: pyrealsense2 get_distance() return depth distance of 0

anguyen216, 2020-02-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | v2.31.0 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| Language                            | python | 
| Segment			|  others | 

### Issue Description
Hi,

I'm trying to calculate distance between two pixels based on their depth distance from the camera.  I'm performing this calculation for a single depth frame of different depth video (which also records infrared frames).  I create a function to calculate the distance between pixels, given their coordinates, the depth frame and the corresponding infrared frame.  The function is included below

```
#!/usr/bin/env python3
# distance.py

import numpy as np
import pyrealsense2 as rs

def pixel_distance(depth_frame, ir_frame, x1, y1, x2, y2):
    """
    Function to compute distance between two depth pixel
    follow the calculate_distance function from this example
    https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

    Inputs:
    - depth_frame: individual intel realsense depth frame
        not in numpy array format
        frame = pipeline.wait_for_frame()
        depth_frame = frame.get_depth_frame()
    - ir_frame: individual intel realsense infrared ir_frame
        not in numpy array format
        frame = pipeline.wait_for_frame()
        ir_frame = frame.get_infrared_frame()
    - x1, y1: pixel coordinates of point 1
    - x2, y2: pixel coordinates of point 2

    Output:
    - distance between two pixels in meter (as depth is measured in meter)
    """

    # get infrared frame intrinsic
    intrin = ir_frame.profile.as_video_stream_profile().intrinsics

    # get pixels' depth
    depth1 = depth_frame.get_distance(x1, y1)
    depth2 = depth_frame.get_distance(x2, y2)

    # project pixels onto 3D plane based on their depth
    # project data onto intrinsics of infrared frame
    p1 = rs.rs2_deproject_pixel_to_point(intrin, [x1, y1], depth1)
    p2 = rs.rs2_deproject_pixel_to_point(intrin, [x2, y2], depth2)

    # compute distance
    dist = np.sqrt(np.power((p1[0] - p2[0]), 2) + np.power((p1[1] - p2[1]), 2) + np.power((p1[2] - p2[2]), 2))

    return dist
```
This code succeed when I tried it on one depth frame of a test video.  I later tried this on a collection of depth frames from different videos.  However, this function return 0 for all frames in the collection except the last frame.

I manually check the frame that return 0 distance and found that the get_distance() function return 0 depth for the given pixels.

Below is the code of how I collect depth frame and compute pixels' depths (frame and pixel locations within each frame are chosen manually)

```
# collect all files
files = glob.glob("../data/pigs/*.bag")
files.sort()

# collections of depth frames and infrared frames
Depths = []
Infrareds = []

for idx, filename in enumerate(files):
    # extract pig id 
    #print(filename)
    pid = filename.split(".b")[0].split("\\")[-1]
    #print(id)
    # for some reasons kernel died halfway through the files
    # setup pipeline for each bag file
    # create pipeline
    pipe = rs.pipeline()
    # create config object
    cfg = rs.config()
    # tell config that we will use recorded device from file
    # to be used by the pipeline through playback
    cfg.enable_device_from_file(filename)
    # start streaming from file
    profile = pipe.start(cfg)
    # setup colorizer for depthmap
    colorizer = rs.colorizer()

    # setup playback
    playback = profile.get_device().as_playback()
    playback.set_real_time(False)
    # get the duration of the video
    t = playback.get_duration()
    # compute the number of frames (30fps setting)
    frame_counts = t.seconds * 30

    # extract depth frame
    # colorize depth frame
    # save colorized depth frame as np array for manual examination
    for i in range(frame_counts):
        frames = pipe.wait_for_frames()
        if i == df['frame'][idx]:
            #print(i)
            depth = frames.get_depth_frame()
            Depths.append(depth)
            ir = frames.get_infrared_frame()
            Infrareds.append(ir)

    pipe.stop()

est_len = []
est_w = []
for i in range(df.shape[0]):
    d = Depths[i]
    ir = Infrareds[i]
    # get pixels of length
    x1, y1, x2, y2 = df.iloc[i, 6:10]
    length = pixel_distance(d, ir, x1, y1, x2, y2)
    # get pixels of width
    x1, y1, x2, y2 = df.iloc[i, 10:14]
    w = pixel_distance(d, ir, x1, y1, x2, y2)
    est_len.append(length)
    est_w.append(w)
```

Can anyone help to spot what I did wrong? And what do I have to do to achieve what I want?

I want to go through a collection of depth frames, each from a different depth video, and compute distance between pre-specified pixels based on their depth distances from the camera

Thanks!

dorodnic, 2020-02-19:
Hi @anguyen216 
When confidence in pixel depth is too low the pixel will be marked 0. This is expected behaviour near object boundaries, since there are areas visible from left sensor but not from right one, or wise-versa. 
This is why provided C++ measure example is using hole-filling filter to handle 0 pixels.

anguyen216, 2020-02-19:
> Hi @anguyen216
> When confidence in pixel depth is too low the pixel will be marked 0. This is expected behaviour near object boundaries, since there are areas visible from left sensor but not from [...]

I tried hole filling filter and I still get depth distance (get_distance) of 0.  Also, the pixel location I choose all have non-zero values; I check this by colorize the depth map and then convert it to numpy array using np.asanyarray() function.  Then I check the color pixel value of the numpy array.  All pixel locations I manually choose have non-zero in at least 2/ out of 3 channels (RGB).

anguyen216, 2020-02-19:
In addition, the get_distance() function seems fickle to me.  I test the same code above on a single bag video and single frame.  Every time I rerun the code, I'd get different answer, in the +/- 3 range of the true value that I expect.  I put the code below; I used the same pixel_distance() function as written above

```
# Setup:
# create pipeline
pipe = rs.pipeline()
# create config object
cfg = rs.config()
# tell config that we will use recorded device from file
# to be used by the pipeline through playback
cfg.enable_device_from_file("../data/distance_test.bag")
#cfg.enable_all_streams()
align_to = rs.stream.infrared
align = rs.align(align_to)
# start streaming from file
profile = pipe.start(cfg)
# setup colorizer for depthmap
colorizer = rs.colorizer()

# setup playback
playback = profile.get_device().as_playback()
playback.set_real_time(False)
# get the duration of the video
t = playback.get_duration()
t.seconds
# compute the number of frames (30fps setting)
frame_counts = t.seconds * 30

# extract and save all depth frames
result = []
spec = 0
ir_frames = []
for i in range(frame_counts):
    #print(i)
    frame = pipe.wait_for_frames()
    #aligned = align.process(frame)
    ir = frame.get_infrared_frame()
    ir_frames.append(ir)
    depth = frame.get_depth_frame()
    #result.append(np.asanyarray(colorizer.colorize(depth).get_data()))
    #tmp = np.asanyarray(colorizer.colorize(depth).get_data())
    result.append(depth)
    #io.imsave("../outputs/distance_test/" + str(i) + ".jpg", tmp)
    #result.append(tmp)
    #if i == 16:
        #spec = np.asanyarray(colorizer.colorize(depth).get_data())
        #color = frame.get_color_frame()
playback.pause()
pipe.stop()

dt = pixel_distance(result[31], ir_frames[31], 384, 363, 708, 401)
print(dt)
```

In addition, whenever I tried to aligned the depth frame and the infrared frame (since I did not record the color frame), it give me this error `Error occured during execution of the processing block! See the log for more info`

Below is the code I used with alignment
```
# Setup:
# create pipeline
pipe = rs.pipeline()
# create config object
cfg = rs.config()
# tell config that we will use recorded device from file
# to be used by the pipeline through playback
cfg.enable_device_from_file("../data/distance_test.bag")
#cfg.enable_all_streams()
align_to = rs.stream.infrared
align = rs.align(align_to)
# start streaming from file
profile = pipe.start(cfg)
# setup colorizer for depthmap
colorizer = rs.colorizer()
# setup alignment
align_to = rs.stream.infrared
align = rs.align(align_to)

# setup playback
playback = profile.get_device().as_playback()
playback.set_real_time(False)
# get the duration of the video
t = playback.get_duration()
t.seconds
# compute the number of frames (30fps setting)
frame_counts = t.seconds * 30

# extract and save all depth frames
result = []
spec = 0
ir_frames = []
for i in range(frame_counts):
    #print(i)
    frame = pipe.wait_for_frames()
    aligned = align.process(frame)
    ir = aligned.get_infrared_frame()
    ir_frames.append(ir)
    depth = aligned.get_depth_frame()
    #result.append(np.asanyarray(colorizer.colorize(depth).get_data()))
    #tmp = np.asanyarray(colorizer.colorize(depth).get_data())
    result.append(depth)
    #io.imsave("../outputs/distance_test/" + str(i) + ".jpg", tmp)
    #result.append(tmp)
    #if i == 16:
        #spec = np.asanyarray(colorizer.colorize(depth).get_data())
        #color = frame.get_color_frame()
playback.pause()
pipe.stop()
```

The for loop would be interrupted by the error

anguyen216, 2020-03-04:
Hi, I was never able to resolve this issue.  Please provide more suggestions

AlphaRalph, 2020-04-16:
Hi, I was facing the exact same problem, but I think I figured it out. Btw it was tricky as hell.

Seems that each file-based frame needs its own configuration, which must be preserved until the frames are processed.
I did someting like:

configs = []
for f,i in zip(files, range(len(files)):
  c = rs.config()
  c.enable_device_from_file(filepath, repeat_playback=False)
  configs.append(c)
  frames = load_single_frame_from_file(configs[i], f)

with

def load_single_frame_from_file(rs_config)":
    pipeline = rs.pipeline()
    profile = pipeline.start(rs_config)
    frameset = pipeline.wait_for_frames()
    streams = profile.get_streams()

    ... do stuff with the streams ...

    pipeline.stop()
    return frames

Hope this solves your Problem too.

RealSenseCustomerSupport, 2020-04-28:
----------------------------------------------

@AlphaRalph Thanks for sharing!
@anguyen216 Did you get chance to try what AlphaRalph provided? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@anguyen216 Any update from your side? Thanks!

anguyen216, 2020-05-18:
Thank you @AlphaRalph for suggesting the fix!  I'll try it 
@RealSenseCustomerSupport Sorry for not being responsive recently, just got out of school work to finally have time working on this project.  I will provide more update within the next 2-3 days.  Thank you!

RealSenseCustomerSupport, 2020-05-19:
----------------------------------------------

@anguyen216 Looking forward to your update. Thanks!

anguyen216, 2020-05-26:
@RealSenseCustomerSupport  and @AlphaRalph I haved tried what was suggested, and it looks like it's working for my application.  Thank you!

RealSenseCustomerSupport, 2020-06-01:
----------------------------------------------

@anguyen216 That's great! Thanks for the update! Close this issue accordingly.

closed 2020-06-01 by RealSenseCustomerSupport