realsenseai/librealsense #5882: Android how to persist sensor.setValue ?

Murded, 2020-02-19:
Hi, I'm currently experiencing an issue when using the following code:

```
if(depthSensor.getValue(Option.VISUAL_PRESET) != 5) 
{
    depthSensor.setValue(Option.VISUAL_PRESET, 5);
    depthSensor.setValue(Option.LASER_POWER, 360f);
}
```

However after I rotate the device the depthSensor options need to be set again.

Perhaps I'm doing something wrong but is there a way that I can do so these values don't have to be set again as it takes a few seconds each time.

dorodnic, 2020-02-24:
Hi @Murded 
Right now the value of `Option.VISUAL_PRESET` does not persist when you destroy and recreate the device object, but if you issue `setValue` during streaming (or you call `start` after `setValue`  before rotating the device) its effects are already applied to the camera. Setting new value to `Option.VISUAL_PRESET` will set many different controls on start, but we don't want to be reading  all these controls every time object is created, hence the option itself resets to custom. 
You only need to reset it when camera is re-connected.

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@Murded Any other questions about this with the reply from dorodnic? Looking forward to your update. Thanks!

Murded, 2020-03-04:
@RealSenseCustomerSupport My appologise I thought I had responded.

@dorodnic Does that mean if I keep a reference to the device (in a static class for example) that I can then re-use that device and if I query ``` Option.VISUAL_PRESET ``` it will have the correct value?

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

Please make sure "You only need to reset it when camera is re-connected."
and only one librealsense app is running at the same time.

RealSenseCustomerSupport, 2020-03-31:
----------------------------------------------

@Murded Any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@Murded Could you please update? Thanks!

Murded, 2020-04-08:
> @Murded Could you please update? Thanks!

You can go ahead and close this

RealSenseCustomerSupport, 2020-04-15:
----------------------------------------------

@Murded Thanks for the feedback! Close this accordingly.

closed 2020-04-15 by RealSenseCustomerSupport