realsenseai/librealsense #5883: Pyrealsense2/T265 cannot use context manager to ask serial number and then connect

MissBeate, 2020-02-19:
Required Info |  
-- | --
Camera Model | T265
Firmware Version | 0.2.0.879
Operating System & Version | Win10
Kernel Version (Linux Only) | (e.g. 4.14.13)
Platform | PC
SDK Version | 2.31.0.1234
Language | python
Segment | {Robot/Smartphone/VR/AR/others }

### Issue Description
Hi!

I'm connecting to two T265 cameras using python and try to connect them both by first asking the serial number off the context manager and then starting the two pipelines.

The program recognises 2 connected cameras and reads their serial numbers just fine. But then when I want to do pipe.start(cfg) it will not work. Giving
RuntimeError: No device connected

On the other hand I can hard-code the serial numbers in cfg.enable_device(xxx) and it works just fine. I can work with this for the moment because I only have those 2 cameras but it would be nice to not have to hard code the serial numbers.

This is the code I use, it gives the error in line 'pipe.start(config)'.
`
realsense_ctx = rs.context()
connected_devices = []
for i in range(len(realsense_ctx.devices)):
    detected_camera = realsense_ctx.devices[i].get_info(rs.camera_info.serial_number)
    connected_devices.append(detected_camera)

pipelines = []
configs = []
for dev in connected_devices:
    pipe = rs.pipeline()
    print(pipe)
    config = rs.config()
    print(config)
    config.enable_device(dev)
    config.enable_stream(rs.stream.pose)
    pipe.start(config)
    pipelines.append(pipe)
    configs.append(config)
`

Thanks,
Beate

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi @MissBeate

Thanks for the issue submission here.  There are known issues around using multiple cameras.

stino78, 2020-03-08:
Dear Intel Realsense,

I have the same problem, runnung Ubuntu 18.04 and the last version of librealsense 2.33.1.1370.

Can you help with that?

Thank you,

Agostino

RealSenseSupport, 2021-01-07:
Hi @MissBeate 

Hopefully the workaround that you've done using the serial number is still working for you.  We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-16 by RealSenseSupport