realsenseai/librealsense #5884: Support for Filter Parameters in NodeJS Wrapper?

prkirby, 2020-02-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { MacOS } | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| Language                            |  {nodejs}                          | 

### Issue Description
It looks like in the Node wrapper, there is no way to send filter parameters to the various filters, as can be seen [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#post-processing-filters)? 

I see that in the `process()` function for filter, on line 3163 of the wrapper's `index.js`, it is checking for arguments, but not doing anything with them. Is this a future feature that is being looked into? Or am I missing something completely?

gavan1, 2020-02-19:
@prkirby have you gotten the nodejs wrapper to actually function?

prkirby, 2020-02-20:
@gavan1 Yeah! It was a bit of a mess to untangle. Make sure you back up to about v10.16 with node (use nvm to make it easy), and if build fails not able to find the realsense.dy file or whatever it is inside of /Release, you just need to make that directory and then set up a symbolic link to the file. Hope that helps!

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

@prkirby Sorry for late response. Just want to check with you if you still need the support for filter parameters? Thanks!

prkirby, 2020-03-19:
Yeah, I'd love to know if thats possible? I had no options for dialing
things in, just had to hope for the best with the default parameters

Regards,

*Paul Kirby*
*Web Developer / Audio Engineer*
*Chicago, IL*

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@prkirby Thanks for the reply! We'll evaluate this and feedback to you later.