realsenseai/librealsense #5886: Follow-up question to #1686

chrelli, 2020-02-20:
No, -DCMAKE_BUILD_TYPE=Release does not fix it. I had already compiled the release version.

Just so that I understand the expected behavior. Is the following correct?

When a frame is dropped, the last known frame, frame number and hardware time stamp is passed (so it's repeated). OK. But when the next new frame arrives, is the hardware time stamp counter is just updated, as if the dropped frame never happened?

I mean, if record at 60 fps, the time difference in the hardware time stamps is ~16 ms.

Let's say that I drop 2 frames, then the difference in time stamps will be like this:
[16, 16, 16, 0, 0, 16, 16, 16, 16]

Even though, in reality, it should be something like:
[16, 16, 16, 0, 0, 48, 16, 16, 16]

Since, in the real world, 3 frame cycles has elapsed while the camera was dropping frames.

I mean, does a dropped frame just mean that the camera misses one cycle, at the same frame rate?

_Originally posted by @chrelli in https://github.com/IntelRealSense/librealsense/issues/1686#issuecomment-588597912_

ev-mp, 2020-02-26:
@chrelli hello, I've read both the original and the current thread but still could not elicit the core claim. I will try to answer inline:
>When a frame is dropped, the last known frame, frame number and hardware time stamp is passed (so it's repeated). OK. But when the next new frame arrives, is the hardware time stamp counter is just updated, as if the dropped frame never happened?

No. The frames and their respective metadata attributes are encapsulated and handled atomically. the SDK does not manipulate the frames content/attributes.
Thus the following scenario does not apply to Librealsense:
>Let's say that I drop 2 frames, then the difference in time stamps will be like this:
~~[16, 16, 16, 0, 0, 16, 16, 16, 16]~~

>I mean, does a dropped frame just mean that the camera misses one cycle, at the same frame rate?

The camera cannot miss a frame, but frame drops may and do occur along the propagation chain : USB PHY/Controller  -  Kernel  -  Application.  So I'm not sure I fully understand the question - can you clarify ?

ev-mp, 2020-03-30:
@chrelli hello, did it answer your question ?
If no further questions we close the ticket.

RealSenseCustomerSupport, 2020-05-02:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-08 by RealSenseCustomerSupport