realsenseai/librealsense #5887: Tips to reduce drift with T265 in Unity

ArvindSanjeev, 2020-02-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (Open RealSense Viewer --> 0.2.0.926) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.32.1 }   | 
| Language                            |  { unity }                          | 
| Segment			|  {AR}                          | 

### Issue Description
Hello,
I am running T265 with Unity for an AR project. I am currently able to create a map of a space (exporting the x,y,z coordinates to a text file) in Unity and then save it, after which I place digital assets (.png's) at different points in the space through Unity so that the next time I walk around that area with the T265, I see that asset in my tablet PC.
[image]

Before starting the scene again, I also place the T265 in a marked origin location in the real world. The problem is that after I start the scene and walk through it, there is a significant amount of drift compared to before (roughly 1-2m). I wanted to check what are the best ways to reduce this drift, I have been searching around and found some potential directions, but none of them have been used inside Unity, such as:
1. Relocalizing based on a map exported earlier. How to do this through Unity?
2. April tags. is this a viable approach to reducing drift and recalibrating? How to do this inside Unity?
What would be the best way to reduce drift and make the tracking stick each time?

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi @ArvindSanjeev

Thanks for this thread.  The tips or ways in which to reduce drift using AprilTag or relocalization does indeed help reduce the drift.  These features are standard in librealsense master branch.  Although I would say the AprilTag is an example and not a native feature of the T265 FW directly, it's a way of tracking using host side processing.

My main point here is that these are a part of librealsense and not a part of the Unity wrapper.  Unfortunately the Unity wrapper feature set is limited.

ArvindSanjeev, 2020-02-26:
Thanks for the response, will try experimenting outside Unity.

closed 2020-04-30 by RealSenseSupport