realsenseai/librealsense #5888: ROS tf for T265+D435 combo mount

FPSychotic, 2020-02-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, I'm using this mount on my robot
https://github.com/IntelRealSense/librealsense/tree/development/examples/tracking-and-depth
Also there is a config file, I guess for Realsense-viewer and SDK, but I can't or I don't how use it as TF on ROS. Could someone please give me the ROS TF between this two cameras or how I can get it? Hand  measuring I guess is not good idea,I don't know use CAD software and I don't even what physical points I need mesure. Should be great add a file too with it (and of the other mount too)
Thank you!!

RealSenseCustomerSupport, 2020-02-22:
----------------------------------------------

Hi @FPSychotic

You can have a look at the following pull request that outlines discussion around the mounting of T265 and D435.  Other than this, we do not have any other information around ROS TF and mounting solution.

https://github.com/IntelRealSense/realsense-ros/pull/764

FPSychotic, 2020-02-22:
Ok should  be great give the ROS TF between both cameras in both Intel mounts, that would help a lot to begginers and no so beginners. As Intel recommend strongly use both realsenses and wheel encoders should be nice a more clear info to make them work.  Thanks

closed 2020-02-22 by FPSychotic

The-Michael-Chen, 2025-02-03:
Does anyone still have the cad file for that bracket?