realsenseai/librealsense #5890: Example of multiple sensors on one camera

HHCBakker, 2020-02-20:
| Camera Model                       | D435 } | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Win7 SP1 (under Mac OS Catana and Parallels 15.1.2)  | 
| Platform                                 | iMac  |
| SDK Version                            |  2.0                          | 
| Language                            |  C                          | 

I'm having trouble getting streams to work from two sensors on the same camera; there doesn't seem to be an example of this configuration here or in the SDK, nor any posts covering the topic. It's not clear whether one has to create two streams or one composite stream. 

I have successfully streamed the colour and the depth separately but I can't get both to work together using one stream for each sensor. I get an "Acquire failed" error when I use the rs2_pipeline_start_with_config function to start the second pipeline (I tried both the colour and the depth stream first).

I have used 640x480 at 6 fps so as to remove the issue of bandwidth.

MartyG-RealSense, 2020-02-20:
The MacOS installation instructions were recently updated to work with Catalina.  So using your D435 with Catalina directly may provide a better experience than running Windows 7 through Catalina.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

I would recommend using 15 FPS as a minimum, as historically there have been errors with 6 FPS streaming that disappear when the FPS is increased to at least 15.

If you are streaming depth and color on the same camera then you can combine them using an Align process.

There is a C++ example:

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

If you are looking for a "C" example (original C, not C++ or C#) as the specification at the top of your message indicates, I do not know of example depth and color alignment code for that language, unfortunately.  One of the RealSense team members on this forum may be able to offer advice on that.

HHCBakker, 2020-02-21:
I'm using the Mac as a development platform but the target is a Win7 PC. 

I've had a look at the Align example and that has answered most of my questions. Set up two streams but only one pipeline. The frames then appear in pairs when I wait for them. I'm not concerned with aligning them as I only want about one frame a second anyway.

The only problem I now have is that, in C, there does not seem to be a good way of determining which frames are depth frames and which are colour frames. C++ has a get_color_frame and a get_depth_frame but C doesn't. I've had to resort to using the stride and the width to infer which is which.

Other than that, I'm happy. Thanks for supplying the (obvious after the fact) solution. 🙂

closed 2020-02-21 by HHCBakker

MartyG-RealSense, 2020-02-21:
You,'re very welcome.  Glad you were able to find a solution!