realsenseai/librealsense #5891: UVC header is not available

MoBaT, 2020-02-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.2.100 | 
| Operating System & Version |  Ubuntu 18.04.4  | 
| Kernel Version (Linux Only)    |  5.3.0-40-generic | 
| Platform                                 | X86_64 |
| SDK Version                            |  2.32.1                          | 
| Language                            |  C++ | 
| Segment			|  | 

I'm having trouble getting getting global timestamps from the SDK. When I attempt to call get_frame_metadata() from the SDK, I get `UVC header is not available`.

What I have done was checkout the "development" branch and run the:
./scripts/setup_udev_rules.sh
./scripts/patch-realsense-ubuntu-lts.sh

According to user @ev-mp, support for kernel 5.3 was added recently #5751 . The kernel modifications were successfully built and applied but the issue still persists.

MoBaT, 2020-02-20:
So, I rebuilt the librealsense2 sdk development branch and used the generated shared objects and it solved my problems.

ev-mp, 2020-02-26:
@MoBaT , thanks for the feedback.
Closing this as resolved. Feel free to reopen if needed.

closed 2020-02-26 by ev-mp