realsenseai/librealsense #5892: getting depths in meters in matlab from bag file

nawara72, 2020-02-20:
Hi

I recorded a bag file using rs-record-playback

i used rs_bag2image to generate the depths images. When I read those images in matlab and get the values for a specific coordinate, I get the correct distance
eg
d=imread('000089.png');
d(256,400)
ans =
  uint16
   1894

How can I achieve the same thing using rs-convert?
I converted the bag to png & raw but i get incorrect values when I load in matlab. What am I doing wrong?
eg
FID=fopen("_Depth_89.raw");
depth=fread(FID,[640,480],'uint16');
depth(256,400)
ans =
     0

Thanks in advance for all your help

MartyG-RealSense, 2020-02-20:
There is a long discussion in the link below about getting depth from a bag in MATLAB.

https://github.com/IntelRealSense/librealsense/issues/3241

I also recall that an Intel RealSense support member once recommended the script below for reading a RealSense raw file in MATLAB.

fid = fopen("FilesName.Raw");
Depth = fread(fid, [1280 720], '*uint16')';
fclose(fid);

nawara72, 2020-02-20:
MartyG-RealSense
Thank you so much for your extremely quick reply. I just have to first say the always quick replies from you guys is amazing, so helpful and very appreciated.

As you see from my question, I already use the method you mention to read the image but somehow I am not getting the correct value. Even though rs_bag2image  is giving me the correct values, I would prefer to use rs-convert instead.

Thanks again for your help

> There is a long discussion in the link below about getting depth from a bag in MATLAB.
>
> #3241
>
> I also recall that an Intel RealSense support member once recommended the script below for reading a [...]

MartyG-RealSense, 2020-02-20:
Yes, I saw that your .raw loading code was mostly similar to the Intel support member's version.  I just wanted to post it so you could compare it to your own as verification.

The author or **rs_bag2image**, UnaNancyOwen,  once said that their program "outputs the original data as it is".  This may account for your successful results with PNGs generated from it.

Another RealSense user provided further explanation about differences between the functioning of rs_bag2image and rs-convert:

https://github.com/IntelRealSense/librealsense/issues/3208#issuecomment-466146575

nawara72, 2020-02-21:
Saw those pages, didn't really help much. Thanks you

Hope someone else has any ideas

Thanks

> Yes, I saw that your .raw loading code was mostly similar to the Intel support member's version. I just wanted to post it so you could compare it to your own as verification.
> 
> The author of **rs_bag2image**, UnaNancyOwen, once said that their program "outputs the original data as it is". This may account for your successful results with PNGs generated from it.
> 
> Another RealSense user provided further explanation about differences between the functioning of rs_bag2image and rs-convert:
> 
> [#3208 (comment)](https://github.com/IntelRealSense/librealsense/issues/3208#issuecomment-466146575)

nawara72, 2020-02-24:
Just figured out, I had the image size wrong, so now all working fine

Thanks MartyG 

> Saw those pages, didn't really help much. Thanks you
>
> Hope someone else has any ideas
>
> Thanks

closed 2020-02-24 by nawara72

MartyG-RealSense, 2020-02-24:
Awesome news, thanks so much for the update!